microsoft/Azure-Kinect-Sensor-SDK #972: Can I get the depth information of each pixel in the RGB image？

luxiansheng96, 2019-12-13:
Hello,Can I get the depth information of each pixel in the RGB image？
I The size of the RGB image I collected using Azure Kinect is 4096 * 3072, and the depth image is 1024 * 1024. But what I need is the depth information corresponding to each pixel on the RGB picture. Obviously, these two sizes cannot match.
This question is very important to me, and I hope to get your help, thank you very much！

RoseFlunder, 2019-12-13:
Hi,
you could use one of these transformations from the SDK
https://docs.microsoft.com/en-us/azure/kinect-dk/use-image-transformation

elk-april, 2019-12-16:
Thanks so much, I've got the depth image I want (same resolution as RGB).
Thanks again for your answer!

closed 2019-12-16 by elk-april