microsoft/Azure-Kinect-Sensor-SDK #973: OpenGL Compatability Issues

R0NAM1, 2019-12-13:
**Describe the bug**

When trying to use the 3d viewer with K4AViewer the program crashes with
`terminate called after throwing an instance of std::logic_error
what(): Shader program linking error:
error: Too many vertex shader image uniforms (1 > 0) 

Aborted (core dumped)`

I believe this is an OpenGL issue because of the next symptom with 3d, when I run k4abt_simple_3d_viewer CPU (As the laptop I am using has only integrated intel graphics) I always get the error of 
'Shader Error: 0:1(10): error: GLSL 4.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, 3.00 ES, and 3.10 ES'

seeing this I set the environment variable 'MESA_GL_VERSION_OVERRIDE' to  'MESA_GL_VERSION_OVERRIDE=3.0', in which I get the error 'OpenGL 4.4 not supported. Please install the latest graphics drivers.'
Trying the value of '4.4' gives a new error: 
'Shader Error: 0:2(38): error: gl_FragColor undeclared
0:2(38): error: value of type vec4 cannot be assigned to variable of type error'

**To Reproduce**
Have the latest intel graphics drivers for Linux and run 3d mode in k4aviewer or k4abt_simple_3d_viewer CPU 

**Expected behavior**

A perfect (laggy) 3 dimensional view of what the camera can see + visible body tracking for k4abt_simple_3d_viewer 

**Logs**

**Screenshots**

**Desktop (please complete the following information):**
 - OS with Version: 
   Manjaro (Kernel 5.4.2-1-MANJARO) (I know Manjaro is not an officially supported OS, but since it is very compatible with Ubuntu it all works the same, the software is just installed differently, plus Ubuntu has the same exact issues here, no differences at all, same driver too!)
 - SDK Version:  
  SDK 1.3.0 + Dev 
 K4ABT 0.9.4 + Dev
 - Firmware version:  
Firmware version: Current Firmware Versions:
RGB camera firmware: 1.6.102
Depth camera firmware: 1.6.75
Depth config file: 5006.27
Audio firmware: 1.6.14
Build Config: Production

**Additional context**

cdedmonds, 2019-12-21:
OpenGL 4.4 is required for Azure Kinect as documented here:
https://docs.microsoft.com/en-us/azure/kinect-dk/system-requirements

You can check Intel's OpenGL support for different chipsets here:
https://www.intel.com/content/www/us/en/support/articles/000005524/graphics-drivers.html

R0NAM1, 2019-12-24:
> 
> 
> OpenGL 4.4 is required for Azure Kinect as documented here:
> https://docs.microsoft.com/en-us/azure/kinect-dk/system-requirements
> 
> You can check Intel's OpenGL support for different chipsets here:
> https://www.intel.com/content/www/us/en/support/articles/000005524/graphics-drivers.html

I am currently running OpenGL 4.5, is it required that I downgrade to only use 4.4?

rabbitdaxi, 2020-01-06:
@R0NAM1 thank you for the feedback. Your issue indeed find a typo in the logging. "OpenGL 4.4 not supported. Please install the latest graphics drivers.", this is incorrect, 4.4 is the minimum supported opengl verison. We will fix the logging.

However, if you see such a message show up, this means the opengl context creation failed. Can you check the OpenGL version with glxinfo and look at the core profile version string and make sure it is 4.4 or higher? one example below is:

Make sure that OpenGL version is 4.4 or higher. Type: glxinfo | grep OpenGL. You should get something like this:
OpenGL vendor string: NVIDIA Corporation
OpenGL renderer string: GeForce GTX 1080 Ti/PCIe/SSE2
**OpenGL core profile version string: 4.6.0 NVIDIA 430.26**
OpenGL core profile shading language version string: 4.60 NVIDIA
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 4.6.0 NVIDIA 430.26
OpenGL shading language version string: 4.60 NVIDIA
OpenGL context flags: (none)
OpenGL profile mask: (none)
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 430.26
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
OpenGL ES profile extensions:

R0NAM1, 2020-01-07:
> @R0NAM1 thank you for the feedback. Your issue indeed find a typo in the logging. "OpenGL 4.4 not supported. Please install the latest graphics drivers.", this is incorrect, 4.4 is the minimum [...]

The Core profile string states I am running 

> `4.5 (Core Profile) Mesa 19.3.1`

rabbitdaxi, 2020-01-07:
@R0NAM1 Just curious, before you run the body tracking viewer (with overriding those environment variable), have you ever successfully run k4aviewer 3d view before at all? It might be good to peel the onion by looking at k4aviewer first, which should work with opengl 4.4 and above.

R0NAM1, 2020-01-08:
> 
> 
> @R0NAM1 Just curious, before you run the body tracking viewer (with overriding those environment variable), have you ever successfully run k4aviewer 3d view before at all? It might be good to peel the onion by looking at k4aviewer first, which should work with opengl 4.4 and above.
> 
> Also, for linux, we only tested on Ubuntu. Please see whether you can also repro with: "Linux Ubuntu 18.04 (x64) with OpenGLv4.4 or later GPU driver" (https://docs.microsoft.com/en-us/azure/Kinect-dk/system-requirements)

Both in Ubuntu and Other Linux Distro's I have never been able to run the K4A viewer in 3d, all the interfaces work in 2d, in 3d though it just crash's with ```
terminate called after throwing an instance of std::logic_error
what(): Shader program linking error:
error: Too many vertex shader image uniforms (1 > 0)

Aborted (core dumped)
```

rabbitdaxi, 2020-01-08:
@R0NAM1 can you also confirm you indeed can run k4aviewer and see 2d depth image on your linux setup? if that is the case, it means the depth engine works fine and your opengl version is okay. 

The crash you mentioned might be from the k4aviewer app.

R0NAM1, 2020-01-09:
> 
> 
> @R0NAM1 can you also confirm you indeed can run k4aviewer and see 2d depth image on your linux setup? if that is the case, it means the depth engine works fine and your opengl version is okay.
> 
> The crash you mentioned might be from the k4aviewer app.

I can indeed, it is the thermal looking stream correct? Red is close, blue is far, black is the void.

rabbitdaxi, 2020-01-09:
@R0NAM1 yes, correct. That is the depth image, along with it, you also should see an IR image in the k4aviewer. The fact you can see these 2d images which are both computed by the libdepthengine on the linux, it means the opengl version that libdepthengine expecting was met. Then for the other error of the "error: Too many vertex shader image uniforms (1 > 0)" when you switch to the 3d view in k4aviewer, it might be something computer resources related but I am not sure. It will be good that you can provide the GPU info here. 

Adding triage needed back to the issue.

R0NAM1, 2020-01-11:
> @R0NAM1 yes, correct. That is the depth image, along with it, you also should see an IR image in the k4aviewer. The fact you can see these 2d images which are both computed by the libdepthengine on [...]

I am using two laptops (both have the exact same problem) so respectively:
Ubuntu: Lenovo G50-80, "Intel HD Graphics 4400"
Manjaro: Dell Latitude 3340, "Intel Integrated HD Graphics"
Thing is for both of these the 3d viewer and body tracker work 'fine', as in take forever to launch and run at the speed of the IBM 7094, but none the less they work.

rabbitdaxi, 2020-01-14:
@R0NAM1 , discussed with the triage team, given there are many different topic/error covered in this issue. Initially this issue was about the OpenGL version, but then it seems you can get 2D depth images successfully but hitting some other error in 3d viewer. You did also mentioned various laptops and various Linux OS as well as two different view. It start to get quite confusing, therefore, we propose the following:
1. Please ensure you read https://docs.microsoft.com/en-us/azure/Kinect-dk/system-requirements (we are only supporting L**inux Ubuntu 18.04 (x64)** with OpenGLv4.4 or later GPU driver). Therefore, when you try to repro, please make sure you have this OS installed and the hard ware meeting the minimum host PC requirements. 
2. Please double check you have OpenGL 4.4 and later with "glxinfo | grep OpenGL". Then you should be able to run **k4aviewer.exe** and see the 2d depth and ir images. Based on above conversation of this github issue, it seems you are reaching to this part. 
3. Now, if above two steps all looking good, and if you hitting issue when you switch to 3d view in the **k4aviewer.exe**, please open a new github issue and provide the error/callstack as well as what depth/color mode you are using from the viewer

closed 2020-01-14 by rabbitdaxi

ChLee98, 2020-03-13:
> OpenGL vendor string: Intel Open Source Technology Center
> OpenGL renderer string: Mesa DRI Intel(R) Haswell Mobile
> OpenGL core profile version string: 4.5 (Core Profile) Mesa 19.3.1
> OpenGL core profile shading language version string: 4.50
> OpenGL core profile context flags: (none)
> OpenGL core profile profile mask: core profile
> OpenGL core profile extensions:
> OpenGL version string: 3.0 Mesa 19.3.1
> OpenGL shading language version string: 1.30
> OpenGL context flags: (none)
> OpenGL extensions:
> OpenGL ES profile version string: OpenGL ES 3.1 Mesa 19.3.1
> OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.10
> OpenGL ES profile extensions:
> 
> This is the output of glxinfo | grep OpenGL, which does show:
> `OpenGL core profile version string: 4.5 (Core Profile) Mesa 19.3.1`
> 
> but it also shows
> 
> `OpenGL version string: 3.0 Mesa 19.3.1`
> 
> I'm guessing this is the issue.

Have you solve the problem? I had the same problem, and at first I run glxinfo | grep OpenGL, which show the same information as you. Finally, I solved this problem by downloading and replacing the nvidia driver. Now, I run glxinfo | grep OpenGL, which shows as follow:
OpenGL vendor string: NVIDIA Corporation
OpenGL renderer string: GeForce GT 740M/PCIe/SSE2
OpenGL core profile version string: 4.5.0 NVIDIA 384.130
OpenGL core profile shading language version string: 4.50 NVIDIA
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 4.5.0 NVIDIA 384.130
OpenGL shading language version string: 4.50 NVIDIA
OpenGL context flags: (none)
OpenGL profile mask: (none)
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 384.130
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
OpenGL ES profile extensions: