microsoft/Azure-Kinect-Sensor-SDK #974: DLL k4a not found

qiufeng2014, 2019-12-13:
1. Project > Manage Nuget Package >  Download Microsoft.Azure.Kinect.Sensor
2. MainWindows.xaml.cs  using kinect=Microsoft.Azure.Kinect.Sensor;
3. Run Code kinect.Device.Open() will error
How to solve it?

Error Information:
--------------------------------------------------------------------
System.DllNotFoundException
   HResult = 0x80131524
   Message = Unable to load DLL "k4a": The specified module cannot be found. (Exception from HRESULT: 0x8007007E).
   Source = Microsoft.Azure.Kinect.Sensor
   StackTrace:
    at Microsoft.Azure.Kinect.Sensor.NativeMethods.k4a_set_debug_message_handler (k4a_logging_message_cb_t message_cb, IntPtr message_cb_context, LogLevel min_level)
    at Microsoft.Azure.Kinect.Sensor.Logger.Initialize ()
    at Microsoft.Azure.Kinect.Sensor.Logger.add_LogMessage (Action`1 value)
    at Microsoft.Azure.Kinect.Sensor.AzureKinectOpenDeviceException.ThrowIfNotSuccess [T] (Func`1 function)
    at Microsoft.Azure.Kinect.Sensor.Device.Open (Int32 index)
    at AI_WPF.MainWindow..ctor () in C: \ Users \ Administrator \ source \ repos \ AI_WPF \ AI_WPF \ MainWindow.xaml.cs: line 45

wes-b, 2019-12-18:
My guess is that you have a cloan or zip of the repo, but you have not built the SDK such that k4a.dll is present on your machine. Please try [building](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/building.md#building-and-dependencies) the SDK.

qiufeng2014, 2019-12-20:
Thank you.
I tried to build the SDK.  And also I tried to Download "Azure Kinect SDK 1.2.0.msi" and installed it.
Adding System Environment Variables and User Environment Variables like
"C:\Program Files\Azure Kinect SDK v1.2.0\sdk\windows-desktop\amd64\release\bin\"
or
"C:\Program Files\Azure Kinect SDK v1.2.0\sdk\windows-desktop\x86\release\bin\"

But same error here.  "System.DllNotFoundException"

wes-b, 2019-12-20:
Please share steps to reproduce this problem. I can't build the project without k4a.dll being present.

wes-b, 2019-12-20:
I would suggest:
* Build the SDK (you might need to build from folder 'build/release')
* Open Azure-Kinect-Sensor-SDK\src\csharp\k4a.sln in Visual Studio 2019
* Set 'Microsoft.Azure.Kinect.Sensor.Examples.WPFViewer' as the default solution.
* Build solution
* Run

closed 2020-01-07 by wes-b