microsoft/Azure-Kinect-Sensor-SDK #976: k4a_record_create() fails for most device configuration color formats

benjamminhansen, 2019-12-15:
**Describe the bug**
No image format device configuration seems to work for the Record API. If `k4a_record_create()`, always `k4a_write_capture()` fails saying unexpected image format.
We need to be able to use the SDK Record API. I have a loop which iterates through the possible color format enumerations (0-8) and tries them with my function for retrieving captures from the camera and recording them to a .mkv file. Here is the code for that:

`int possible_formats[4] = { 0,1,2,3 }; // I've tried 4-8 and none of those formats are valid for k4a_record_create()`
`for (const int format_id : possible_formats)`
`{`
	`// Set the confi.color_format to a plausible value, where the x is in that plausible range!`
	`config.color_format = (k4a_image_format_t)format_id; // cast the integer to type enum so the types match`
	`printf("\nTrying enum: %d\n", format_id);`
	`make_save_recording(device, config); // try to make a recording`
}`

1 of 2 things happen:
Either the `k4a_record_create()` succeeds and then I fill an array of captures, which also succeeds. But then when I try to write the captures to the record handle with this code:
`k4a_record_write_capture(record_handle, image_array[capture_index])`
which produces this error for every capture image:

`[2019-12-13 12:17:16.250] [error] [t=11280] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\record\sdk\record.cpp (661): k4a_record_write_capture(). Tried to write capture with unexpected image format.`

Or... the `k4a_record_create()` fails and says:

`[2019-12-13 12:17:16.303] [error] [t=11280] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\record\sdk\record.cpp (86): k4a_record_create(). Unsupported color_format specified in recording: 3`

**To Reproduce**

1. Create configuration of device like so
`k4a_device_configuration_t config = K4A_DEVICE_CONFIG_INIT_DISABLE_ALL;`
`config.depth_mode = K4A_DEPTH_MODE_NFOV_UNBINNED;`
`config.camera_fps = K4A_FRAMES_PER_SECOND_30;`
`config.color_format = K4A_IMAGE_FORMAT_COLOR_MJPG; // or enumeration 1 or 2` 
`config.color_resolution = K4A_COLOR_RESOLUTION_1080P;`
2. Call the k4a_record_create() giving it the device, the device config, some filename, and an empty record handle
 `k4a_record_create("try_this.mkv", device, device_config, &record_handle)`
This should succeed if 3 argument configuration for image format is accepted...
4. Write the header for the record file
`k4a_record_write_header(record_handle)`
3. Create captures as desired. We created an array of captures from the device, one after the other. Then after filling in the array with captures, we access each one and try to record them to the record file.
`k4a_record_write_capture(record_handle, image_array[capture_index])`
We have checked to make sure the record handle succeeded, the captures succeeded, and the header succeeded. The only issues is try to write the capture to the record, like shown.
4. See error: k4a_record_write_capture(). Tried to write capture with unexpected image format.

**Expected behavior**

I hoped since the captures succeeded and the `k4a_record_create()` seemingly accepted the device configuration that the captures would successfully be written to the record file.

**Logs**
Logging documentation is vague. I need an API to not only log errors
I find, but set the file logging or STDOUT logging the environment variables control as they would for a command line running of the executable. 

**Screenshots**

Trying different image formats. 0-3 seem to work.
[image]

Trying to create a record
[image]

Trying to write the header
[image]

Trying to save captures
[image]

Everything works up to trying to save the captures. Then it gives the above defined errors.

**Desktop (please complete the following information):**
 - OS with Version: Windows 10
 - SDK Version:  1.3.0
 - Firmware version:  
[image]

**Additional context**

benjamminhansen, 2019-12-17:
I thought it also might be how I'm reading captures, but I can verify they show items like depth, color, and IR data. So the captures are working, just writing the captures is not. I keep getting the "unexpected image format" error.

wes-b, 2019-12-17:
This error:

> [2019-12-13 12:17:16.303] [error] [t=11280] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\record\sdk\record.cpp (86): k4a_record_create(). Unsupported color_format specified in recording: 3

Means you are trying to save K4A_IMAGE_FORMAT_COLOR_BGRA32 to the MKV which is not supported in our 1.3 release. We have recently added support for BGRA32, and you can get it by building our development branch, but it is not yet released. If you record in BGRA32, you will need a machine with an SSD to handle the needed file data volume. We recommend one of the other file formats.

This error:
>[2019-12-13 12:17:16.250] [error] [t=11280] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\record\sdk\record.cpp (661): k4a_record_write_capture(). Tried to write capture with unexpected image format.

Means that the image format type of the image being written is not the same as the image type used when creating the recording. The format type must remain constant throughout the recording.

benjamminhansen, 2019-12-17:
@wes-b Thank you so much. That makes a lot of sense. 

Now the k4a_record_create() is blocking (it never returns and lets me do things past its line). I have a camera started with a valid configuration. Do I not need to start the cameras? Does the record create method start the cameras? Would a certain configuration for the device ever cause this?

This line of code doesn't ever stop processing and I see no print statements like the failure or success.
[image]

benjamminhansen, 2019-12-17:
Never mind. It was because my file path was not complete. I needed to include the whole path like, "C:/Users/..." and so forth to a valid directory and new filename.

closed 2019-12-17 by benjamminhansen