microsoft/Azure-Kinect-Sensor-SDK #977: Trying to extract IR returns from raw point-cloud footage

Whity220, 2019-12-16:
I am learning to use the Kinect and am wanting to extract the IR returns from raw point cloud footage. I am also trying to find documentation discussing the spectral resolution of the  Kinect, ie the wavelengths that the sensor captures in. I haven’t found anything related to its spectral capability in any official documentation. Has anyone else found any information regarding this element of the Kinect?

Cheer Andrew

wes-b, 2019-12-17:
Not sure what you mean by `returns from raw point-cloud footage`. 

Please check out this page(https://docs.microsoft.com/en-us/azure/Kinect-dk/depth-camera) and this [section](https://docs.microsoft.com/en-us/azure/Kinect-dk/hardware-specification#depth-camera-supported-operating-modes).

The depth mode K4A_DEPTH_MODE_PASSIVE_IR is the only mode that return ambient IR. All other modes filter out ambient IT and only provide IR reflected by the depth camera's IR lasers.

rabbitdaxi, 2019-12-18:
@Whity220 if you want to get the IR image along with the depth image, you can do so. take a look this example code line https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/61a42c77aaf61df8733bed10e20ccbc3a51a6c95/examples/streaming/main.c#L89

Good feedback on the missing wavelengths documentation, add documentation tag to this issue.

wes-b, 2019-12-18:
@rabbitdaxi the wavelength is covered [here](https://docs.microsoft.com/en-us/azure/Kinect-dk/hardware-specification#depth-camera-supported-operating-modes). Does it need to be anywhere else?

Whity220, 2020-01-05:
@wes-b 
Sorry for the miscommunication, I have recorded a 5 second raw mkv file and I wish to view/extract the IR footage from that recorded file.  Is there anyway this can be done?

Thanks for your help.

wes-b, 2020-01-07:
The playback API's will provide you with the same captures you recorded, check out the example here: https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/examples/playback_external_sync. Assuming you recorded IR images, you can use them exactly the same as you could using the Kinect directly.

closed 2020-01-14 by wes-b