microsoft/Azure-Kinect-Sensor-SDK #978: cv::Mat to k4a_image_t transform

elk-april, 2019-12-16:
Can I convert the image from CV ::mat to k4a::image?
Whether to use the k4a_image_create_from_buffer() or other?

wqs111000, 2019-12-16:
```
//c++ code 
//k4a::image img
if (img.get_format()==K4A_IMAGE_FORMAT_COLOR_BGRA32){
//convert the image from k4a::image to CV ::mat
cv::Mat cImg = cv::Mat(img.get_height_pixels(), img.get_width_pixels(), CV_8UC4, img.get_buffer());
//convert the image from CV ::mat to k4a::image
memcpy(img.get_buffer(), &cImg.ptr<cv::Vec4b>(0)[0], cImg.rows*cImg.cols  *sizeof(cv::Vec4b));
}
```

elk-april, 2019-12-16:
> ```
> //c++ code
> //k4a::image img
> if (img.get_format()==K4A_IMAGE_FORMAT_COLOR_BGRA32){
> //convert the image from k4a::image to CV ::mat
> cv::Mat cImg = cv::Mat(img.get_height_pixels(), [...]

thank you for your answering !
I have solved this problem by:
cv::Mat depth = cv::imread(name_depth, IMREAD_UNCHANGED);
depthImage = k4a::image::create_from_buffer(K4A_IMAGE_FORMAT_DEPTH16,1024,1024,1024 * 2,
depth.data,1024 * 1024 * 2,NULL,NULL);

closed 2019-12-16 by elk-april

ChristopherRemde, 2020-07-01:
Hey guys,

Thank you for your code samples!
For my program I also need to convert a BRGA Color Image from cv::Mat to k4a_image_t. Unfourtunately I'm a total beginner regarding C++ and can't get my function to work It throws an Access violation, so I'm probably doing something wrong regarding the creation of the k4a_image buffer. Can anybody help out a newbie?

```

k4a_image_t AzureKinectCapture::opencv_to_k4a_image(const cv::Mat cImg)
{

	k4a_image_t newImage = NULL;

	k4a_image_create(K4A_IMAGE_FORMAT_COLOR_BGRA32, cImg.cols, cImg.rows, cImg.cols * 4 * (int)sizeof(uint8_t), &newImage);

	memcpy(k4a_image_get_buffer(newImage), &cImg.ptr<cv::Vec4b>(0)[0], cImg.rows * cImg.cols * sizeof(cv::Vec4b));

	return newImage;

}
```

ShravyaB, 2020-11-18:
It is possible you are reading a normal RGB image. It has only 3 channels unlike Kinect which gives us an empty 4th channel. That is why you cannot copy it as a block. You need to insert these gaps.
You can convert the format of your opencv Mat from BGR to BGRA using openCV's cvtColor and then call the above function

ChristopherRemde, 2020-11-18:
Hi Shravya,

thank your for your answer! I just checked my code and the OpenCV-Mat I created and, was created with 4 channels:

`cv::Mat cImg = cv::Mat(k4a_image_get_height_pixels(colorImage), k4a_image_get_width_pixels(colorImage), CV_8UC4, k4a_image_get_buffer(colorImage));`

I solved the problem by **not** putting the code lines for converting the CV-Mat to k4a_image inside it's own function but instead putting them inside the same function where I created the CV Mat. It somehow works out then.

If anybody is interested, you can check out the working code here: https://github.com/MarekKowalski/LiveScan3D/blob/AzureKinect/src/LiveScanClient/azureKinectCapture.cpp