microsoft/Azure-Kinect-Sensor-SDK #981: Camera display and Body tracking rate dropped after installing SDK 0.9.5

soocheolyoon, 2019-12-16:
**Describe the bug**

Hi
I was using SDK 0.9.2.
Simple modified simple_3d_viewer example to extract joint information to csv for every image capture from the camera while displaying in window_controller_3d in real time.
In 0.9.2, mean rate for joint detection is about 18 counts/s .
I didn't change camera configuration, so it must be K4A_DEPTH_MODE_NFOV_UNBINNED 30fps.

Recently I updated to SDK 0.9.5, and download samples for 0.9.5 .
Again using simple_3d_viewer, by default, so it must be K4A_DEPTH_MODE_NFOV_UNBINNED 30fps, the rate dropped to 5~6 counts/s.
and I can see that image capture rate itself decreased to those levels.
I tested other options with manual input of "deviceConfig.camera_fps = K4A_FRAMES_PER_SECOND_30;" -
K4A_DEPTH_MODE_NFOV_2X2BINNED : 18 counts/s
K4A_DEPTH_MODE_NFOV_UNBINNED :  5~6 counts/s
K4A_DEPTH_MODE_WFOV_2X2BINNED : 5~6 counts/s
K4A_DEPTH_MODE_WFOV_UNBINNED : the viewer failed to start

**To Reproduce**

Just add body tracking counter to sample code.
Change device configuration, and run.

**Expected behavior**

18 counts / s for all options.

**Logs**

**Screenshots**

**Desktop (please complete the following information):**
 - OS with Version:  Windows 10 Pro
 - SDK Version:   sensor 1.3.0 body tracking 0.9.5
 - Firmware version:  
Current Firmware Versions:
  RGB camera firmware:      1.6.102
  Depth camera firmware:    1.6.75
  Depth config file:        6109.7
  Audio firmware:           1.6.14
  Build Config:             Production
  Certificate Type:         Microsoft

and I'm using i7-8850H 32GB ram, and RTX2080 6GB 
**Additional context**

wes-b, 2019-12-16:
Can you confirm you are building a retail/release build? There are performance issues with debug builds.

yijiew, 2019-12-17:
@soocheolyoon : 18 FPS to 5 FPS seems to be a drastic slow down, especially for your high-end GPU. We did a performance comparison between the 0.9.5 SDK vs. 0.9.2. The performance was very similar. As @wes-b mentioned, could you check whether the build setting is the same between your two builds? If so, it would be great if you could share your binaries for us to further debug.

soocheolyoon, 2019-12-17:
@wes-b @yijiew : wow Thanks a lot. I missed to check that.
After I fixed to release mode, the results are even better than before (as it supposed to be)-
K4A_DEPTH_MODE_NFOV_2X2BINNED : 30 counts/s
K4A_DEPTH_MODE_NFOV_UNBINNED : 20~30 counts/s
K4A_DEPTH_MODE_WFOV_2X2BINNED : 29~30 counts/s

Thank you for help!

@yijiew : It was Release mode when I was using SDK 0.9.2.

closed 2019-12-17 by soocheolyoon