microsoft/Azure-Kinect-Sensor-SDK #983: The SDK's remap undistortion result is different from OpenCV remap

Guptajakala, 2019-12-17:
I have checked the SDK's example here https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/examples/undistort/README.md
It says we can also use OpenCV's remap function and it is faster. So I tried OpenCV remap. Here is how I did.

```
newK, _ = cv2.getOptimalNewCameraMatrix(K, D, (W,H), 1)
map1,map2 = cv2.initUndistortRectifyMap(K, D, np.eye(3), newK, (W,H), cv2.CV_32FC1)
bgr = cv2.remap(bgr, map1, map2, cv2.INTER_NEAREST)
depth = cv2.remap(depth, map1, map2, cv2.INTER_NEAREST).astype(np.float32)
mask=depth>0.1
vs, us = np.where(mask)
zs = depth
xs = ((us - K[0, 2]) * zs) / float(K[0, 0])
ys = ((vs - K[1, 2]) * zs) / float(K[1, 1])
....   #Some code to generate point cloud from xs,ys,zs
```

However, in this way, the resulted point cloud looks different from what I get using SDK's provided point cloud. Is there any thing wrong here? Many thanks!

jasjuang, 2019-12-17:
Which version of OpenCV are you using? There's a pretty recent merge https://github.com/opencv/opencv/pull/15491 that's relates to `cv::undistort`, and it does alter the behavior slightly.

Guptajakala, 2019-12-17:
@jasjuang 
This is the output from conda list
`opencv-contrib-python     4.0.0.21                 pypi_0    pypi`

I will update opencv and test again.

Guptajakala, 2019-12-17:
@jasjuang 
After a python package update, I tested again. 
`opencv-contrib-python     4.1.2.30                 pypi_0    pypi`

Unfortunately, the result is the same. I can share my data with your team privately to help reproduce. It cannot be disclosed publicly here due to some reasons. Could you give an email address that is convenient for you guys?

jasjuang, 2019-12-17:
I never used the undistortion function from Azure Kinect SDK and have always been using OpenCV. I think you can just capture something random with your Azure Kinect, and then show the maintainers the difference between two results. You don't even need to convert the depth image to point cloud, simply provide the undistorted depth image with both methods and the diff between them.

Guptajakala, 2019-12-17:
@jasjuang Oh, I previously thought you were one of the maintainers. It's really nice of you to have replied to me.

Actually, I have been using OpenCV's undistortion until recently I used SDK's API and found there is some difference on the point cloud I get, though the difference is minor.  The undistorted RGB and depth image from openCV all looks rectified. But the point cloud converted from rectified depth using a pinhole camera model is not same as the point cloud using SDK's depth_to_pointcloud API.

It might not be sufficient to just observe the depth image since I'm not sure if it is caused by the depth to point cloud step.

Guptajakala, 2019-12-18:
I finally switched to opencv project point function.

closed 2019-12-18 by Guptajakala