microsoft/Azure-Kinect-Sensor-SDK #984: What is the SDK minimum specification?

devLupin, 2019-12-17:
[image]
I ran 'Azure Kinect Body Tracking Viewer'.
The symptoms are shown in the picture above.

The black window goes off right after it pops up.

I only know the requirement is windows 10.
Should I have an nvidia graphics card? I use cpu embedded graphics.

My developing environment :
O/S : Windows 10
CPU : i7-3770K
Graphic card : embedded(intel HD graphic 4000)
RAM : 16GB(8GB * 2)

Is there a lack of specifications?

devLupin, 2019-12-17:
Sry, I found the body tracking requirements to be more stringent.

The documentation states that this is a 7th generation i5 processor or higher (quad core 2.4GHz).
It's old, but the CPUs are i7 and 3.50 GHz. Does the CPU specification also matter?

thank you.

wes-b, 2019-12-18:
GPU is also very important for body tracking. Requirements are [here](https://docs.microsoft.com/en-us/azure/kinect-dk/system-requirements#body-tracking-host-pc-hardware-requirements)

closed 2019-12-18 by wes-b