microsoft/Azure-Kinect-Sensor-SDK #985: Get current device input bandwidth usage as fraction from usb controller potential.

barakooda, 2019-12-17:
Hi there,
It will be nice to have current USB bandwidth usage per device and the usb controller potential.

Adding data information available for the user in IDS cameras as example:
[image]

tesych, 2020-01-06:
@barakooda thank you for the idea! Please submit this idea on our [customer voice idea site](https://feedback.azure.com/forums/920053) so other users can vote for it as well.

closed 2020-01-06 by tesych

barakooda, 2020-02-09:
I did some [bandwidth tests](https://docs.google.com/spreadsheets/d/1jUg0ub4Il71KLB0mw8_A6ETegq4_5QSh-cBz0DE8beU/edit?usp=sharing) 

It look like there is enough room for 12mp color camera at 30fps ( if bandwidth is the only issue).