microsoft/Azure-Kinect-Sensor-SDK #986: Look like bad aligment between IR to color camera projected space.

barakooda, 2019-12-17:
Any idea about the source of this behavior ?
Thanks. 
https://forum.derivative.ca/t/look-like-bad-aligment-between-depth-to-color-camera-projected-space-b33840/114382

wes-b, 2019-12-18:
See https://docs.microsoft.com/en-us/azure/Kinect-dk/use-image-transformation.

This is a byproduct of the occlusion created by the depth camera. This is also one of reasons we suggest working on the color camera's domain rather than the depth camera domain.

rabbitdaxi, 2019-12-18:
@barakooda to clarify, the occlusion is naturally causing by the placement of depth camera and color camera. To solve occlusion, per pixel depth information is needed. 
- k4a_transformation_depth_image_to_color_camera function handles occlusion. Recommend to use this function. 
- k4a_transformation_color_image_to_depth_camera does not because of it require another pass of warping depth to color which cost compute.

barakooda, 2019-12-18:
Thanks for the quick replies !

closed 2019-12-21 by cdedmonds