microsoft/Azure-Kinect-Sensor-SDK #987: External trigger issue after firmware update 1.6.987014 => 1.6.102075014

markobe-msft, 2019-12-17:
**Describe the bug**

After firmware update 1.6.987014 => 1.6.102075014 external trigger works on at the first run after AK has been plugged into a PC; fails on the next runs as long as slave AK is power plugged. 

When reverted firmware to 1.6.987014 trigger works on every run. 
I've tried 3 devices, all show same bahaveor. 

**Desktop (please complete the following information):**
 - OS with Version:  Windows 1909
 - SDK Version:  1.3.0
 - Firmware version:  
__Issue shows with__
  RGB camera firmware:      1.6.102
  Depth camera firmware:    1.6.75
  Depth config files: 6109.7 5006.27
  Audio firmware:           1.6.14
  Build Config:             Production
  Certificate Type:         Microsoft
  Signature Type:           Microsoft
__No issues with__ 
  RGB camera firmware:      1.6.98
  Depth camera firmware:    1.6.70
  Depth config file:        6109.7
  Audio firmware:           1.6.14
  Build Config:             Production
  Certificate Type:         Microsoft

wes-b, 2019-12-17:
Thanks @markobe-msft! Can you please share repro steps? We are testing this regularly and have not seen this.

markobe-msft, 2019-12-18:
This one is a bit tricky to make exact repro as my own device is triggering AK. I will check what I can do.

wes-b, 2020-01-07:
If you come up with repro steps we can reopen and take a look.

closed 2020-01-07 by wes-b