microsoft/Azure-Kinect-Sensor-SDK #988: Inaccurate camera internal parameters?

rlczddl, 2019-12-17:
Hi, I use a single azure kinect for reconstructing the human body. I use d k4a_transformation_depth_image_to_color_camera() to align the depth map to the color map, used the internal parameters of the color camera to convert the depth map to a point cloud, then rigidly registerd the tranformed point cloud, and finally used Poisson reconstruction algorithm to reconstruct the mesh. Now for texturing maping, I used the rotation matrix of registration and color camera internal parameters to project each face on the reconstructed mesh onto the color image to get the UV coordinates. As a result, I find that there is a large offset between the texture and the mesh. Maybe the camera's internal parameters are not accurate enough?

xthexder, 2019-12-17:
Are you seeing the same offset when viewing the colorized point cloud in Azure Kinect Viewer?
This should tell you if there's a problem with the device, or if there's an issue with how you're handling the texture mapping.

rabbitdaxi, 2019-12-18:
@rlczddl a suggestion to help you narrow down the issue, you may want to find some feature point from the 2d color image that you know, then once you have the point cloud in your final coordinate system, before you do any meshing, try transform/project those corresponding feature points back to the 2d color image to see whether the reprojection error is small. 

When you say use the internal parameters of the color camera to convert the depth map to a point cloud, what do you mean by that? by the way, you can use this function [k4a_transformation_depth_image_to_point_cloud()](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/group___functions_ga7385eb4beb9d8892e8a88cf4feb3be70.html#ga7385eb4beb9d8892e8a88cf4feb3be70) to do so by passing the camera parameter with K4A_CALIBRATION_TYPE_COLOR

rlczddl, 2019-12-19:
Thank you for your suggestion, I will try and then record it here.

rlczddl, 2019-12-19:
Well, the camera parameters are accurate. The problem is that I had a problem calculating the UV coordinates.

closed 2019-12-19 by rlczddl