microsoft/Azure-Kinect-Sensor-SDK #989: k4a_record_create() shouldn't never return

benjamminhansen, 2019-12-17:
**Is your feature request related to a problem? Please describe.**

Yes. It is a bug with k4a_record_create(). When I gave it a bad filename like "try.mkv" instead of "C:/Users/Me/Desktop/try.mkv" it never returned from running. It blocked up or something and I couldn't see any output for what went wrong. It took trial and error to figure out my file name was bad.

**Describe the solution you'd like**

If the file path is bad, and error or something useful should be displayed or returned, not forever loop kind of block which lead me to just guess what was happening.

**Describe alternatives you've considered**

Not Applicable.

**Additional context**

xthexder, 2019-12-17:
A filename input of "try.mkv" should create a recording file in the current directory, and should be valid.
There is no loop or retry logic inside of `k4a_record_create()`, so it should always return success/failure.

Are you able to break into a debugger to see where the function is hanging?

benjamminhansen, 2019-12-18:
@xthexder I broke into a debugger on this line. I tried "step into" but it couldn't find the file with path: d:\a\1\s\extern\azure-kinect-sensor-sdk\src\record\sdk\record.cpp

So it seems as though I don't have access to that file. Its using it for sure though, I just don't know where it is and the debugger couldn't find it either.

So I stepped over instead of into, about 3 times, and I got these errors, but it didn't return out or let me proceed to my next lines (which was a print for failure or success of k4a_record_create()).

[image]

benjamminhansen, 2019-12-18:
Also side question. Would you suggest I create an array of captures first all at once, and then write them to a record all at once? Meaning I get all the captures I need for the recording before writing any of them to a file, and then I write all of them to the file. Or... should I get a capture and save it to a file in one step, and do that over and over again for as many captures as I want?

benjamminhansen, 2019-12-18:
Interesting. I didn't realize the captures retrieved would remain in some queue handled by the camera, and that a copy of the capture was created for my capture object. I kept getting this error:

[image]

And because I set the wait time to infinite it never stopped/returned, just kept showing this error.

So putting the captures in array and then writing them to a file doesn't seem to work. I needed to get a capture and save it to the record file in one breadth.

wes-b, 2019-12-18:
@benjamminhansen the file `record.cpp` is part of this repo. So you do have access. The best way to use a debugger would be to build this SDK yourself and use your built binaries with the debugger.

Our design is for you quickly perform work based on the image (if you need to) and then hand it to the recorder as quickly as possible. The error message about releasing a capture early due to a full queue is because captures are not ready fast enough prevent the queue from filling up. So you do not need the array.

xthexder, 2019-12-18:
I'll need more info than just `record\sdk\record.cpp`, since the whole `k4a_record_create()` function is within that file, and I haven't been able to reproduce this locally.
Ideally you can provide the line number within record.cpp, and the call stack for where the function is hanging.

The official releases contain symbols, so the debugger should be able to load those without building from source.
If you clone the SDK at the same version as you have installed, then you can point the debugger at it to view source, though I don't think this is required to get line numbers.

As for the dropped captures, you may want to take a look at the existing recording tool to see how it handles them:
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/tools/k4arecorder/recorder.cpp#L141
Depending on what you're trying to do, you may be able to just use k4arecorder directly.

benjamminhansen, 2019-12-19:
Well it worked this time with a filename of "try.mkv". I don't know why, but it didn't hang.

benjamminhansen, 2019-12-19:
Thank you for your help. This was very helpful!!

closed 2019-12-19 by benjamminhansen