microsoft/Azure-Kinect-Sensor-SDK #992: I want to connect four Azure kinect to one computer

devLupin, 2019-12-18:
Sorry. I needed a correct answer, so I wrote a question here.

There are four sensors. I want to connect all four computers to one computer. My task is to do skeletal tracing in C ++ / C # (.Net Framework).

1. Provide minimum hardware requirements for working with 4 sensors and 1 computer (CPU, GPU, RAM, etc.).
. 2. The advantage of azure kinect is that you can connect multiple sensors to a single computer. Is it really possible?
3. If I connect multiple sensors to one computer, can there be errors in the joint tracking values?

Please tell exactly because I'm working on a company project

wes-b, 2019-12-18:
Is this the same question as #984

devLupin, 2019-12-19:
@wes-b 
Similar but question number1 was added.
** Connect four Azure Kinect to one Computer.

I think, 1: 1 and 1: 4 connections will be different, and will likely require higher specifications.
Let me know its specification.

Also, when comparing four connections and one connection, does comparing the value extracted from each Kinect increase the error rate?

qm13, 2019-12-19:
@devLupin The short answer is yes you can connect multiple Kinects to one computer and run multiple instances of body tracking within your application. Note however that we do not provide APIs to fuse the multiple tracked skeletons into a single (hopefully more accurate) skeleton. We do provide smoothing control and confidence indicators (per joint) which you can use to support your own fusion.

We have had 8 Kinects connected to a single computer and heard of customers with as many as 12. The number of Kinects you can connect to a single computer really depends on your bandwidth requirements per device and the number of USB controllers in your PC.

The number of instances of body tracking really depends on the power of your NVIDIA GPU. Body tracking today requires ~150 GFLOPS atFP64 and ~1.9 GB of VRAM. You will likely find the VRAM to be the limiting factor. This is guidance only. You really need to experiment with you application to understand exactly how much compute you need.

devLupin, 2019-12-20:
@qm13 

"Note however that we do not provide APIs to fuse the multiple tracked skeletons into a single (hopefully more accurate) skeleton."
In the above part, I think Intel provides a calibration method. I am going to calibrate the camera through calibration and extract the values ​​through KNN algorithm. 
Please let me know if there is a better way than this.

The bottom line is you have to buy and experiment with a decent GPU? 
You can't buy a lot of GPUs for experiments because the company buys them ㅠㅠ......
If you have any more results, please let me know.

Thank you so much.

qm13, 2019-12-21:
Please take a look at the Azure Kinect green screen example https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/examples/green_screen. This examples uses two cameras and includes calibrating the cameras.

You can also use the gpu_device_id parameter in k4abt_tracker_configuration_t struct to invoke trackers on different (smaller ) GPUs.

closed 2020-01-06 by qm13

g212, 2020-10-16:
@devLupin 
@qm13 

Thank you @devLupin for sharing this question. I was wondering if you were able to find the solution for this challenge that needs better documentation from Microsoft. 

Did you find the right pc requirements (GPU, CPU, RAM...) for connecting the 4 cameras?
Were you able to create your tracking system with the 4 cameras? 

I will truly appreciate your support here since I found myself in the same position that you were.

devLupin, 2020-10-18:
@g212 

The only way I came up with is to use threads.

As a result, it's impossible for me.

My system is real time system.
Some of the data was empty. So, the desired result was not produced.

In the end, I configured one camera on one PC.

g212, 2020-10-19:
Thanks @devLupin for your reply! I'm sorry to hear it didn't work. 

Hi @qm13, could Microsoft help us with more guidance around this issue? Currently, you advertise the solution to allow a smooth integration between cameras, but it's clear that your developers could use your support. Buying and testing GPUs is not viable for most of us, but you do have the resources.

jesusatgar, 2020-12-02:
Any updates on this? I'm really interested in multiple camera body tracking

adri1177, 2020-12-10:
Hello, I am also interested in multiple camera body tracking and I would like to know if I need to go to 1 host: 1 camera or I can use a computer similar to this https://www.zotac.com/es/product/mini_pcs/magnus-en52060v to go 1 host: 2/4 cameras.
can someone tell us if they manage to control 4 or more cameras at a good fps with a single host?
@jesusatgar @g212 @devLupin @qm13

alitokur, 2021-08-06:
> @devLupin The short answer is yes you can connect multiple Kinects to one computer and run multiple instances of body tracking within your application. Note however that we do not provide APIs to [...]

dear @qm13 how did you plug multiple kinects into one computer? do you have any special hardware?

Ahmed-KASAPBASI, 2021-09-30:
I have the same question but for kinect v2, I want to connect 2 or 3  Kinect v2 to one Computer?