microsoft/Azure-Kinect-Sensor-SDK #995: Spread  depth engine and other ML process work over more gpus .

barakooda, 2019-12-19:
Hi there,
I`m working with rtx 2080ti with 4 azures and the gpu can easily become saturated 
( as it seem in my setup).

There is option to utilize something like NVlink or multi GPU setup to spread the depth engine and other ML process work over more cores? 

In the docs you mentioned the option to use small external process unit ( like NUC ) per kinect or two,
I try to avoid this option because dealing with aggregation and sync of the data can 
be a big pain and add latency and generally complicating the overall system.

Any plans/tips/solutions and upgrades tackle this topic will be great.

RoseFlunder, 2019-12-19:
I think you could create multiple body trackers with the sdk because in 0.9.5 you can specify which GPU the tracker will use.

tesych, 2020-01-06:
@barakooda did the update described in the above comment solved your problem?

closed 2020-01-06 by wes-b