microsoft/Azure-Kinect-Sensor-SDK #1000: what is the frequency of tof

TxT1212, 2019-12-20:
**Describe the bug**

**To Reproduce**

**Expected behavior**

**Logs**

**Screenshots**

**Desktop (please complete the following information):**
 - OS with Version: 
 - SDK Version:  
 - Firmware version:  

**Additional context**

wes-b, 2019-12-20:
What do you mean by modulated signal frequency? The sensor uses a 850nm laser and there is some additional information here: https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/868#issuecomment-552621616.

TxT1212, 2019-12-22:
> What do you mean by modulated signal frequency? The sensor uses a 850nm laser and there is some additional information here: [#868 (comment)](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/868#issuecomment-552621616).

Thank you for this information! Truly appreciate it.
 Kinect v2 work in 3 modulation signal frequencies，as far as I know. Please correct me if I am wrong. so what about azure kinect?

wes-b, 2020-01-14:
Azure Kinect does not use modulation frequencies at or near 75 MHz or 100 MHz.  We unfortunately cannot make claims about the degree of interference experienced by an external party product as we have no insight into the exact hardware and algorithm implementation. Multiple Azure Kinects can be connected and configured to avoid interference with each other using the sync in/out ports.

closed 2020-01-14 by wes-b