microsoft/Azure-Kinect-Sensor-SDK #1001: Azure Kinect Viewer - Failed to start microphone

rfilkov, 2019-12-20:
**Describe the bug**

Everything worked fine until recently. Then I updated to Windows 10 v1909 (build 18363.535). After that, each time I start Azure Kinect Viewer, open the sensor and attempt to start cameras, I get these two error messages: "Failed to start microphone: Unable to open device!" and then "Failed to create microphone listener: (no error)". 

All cameras (color, depth, IR) work as expected, as you can see. The microphone array is recognized in 'Device manager' under 'Audio inputs and outputs'.

**Expected behavior**

Do you have any idea, why do these two messages pop up after the Windows update?

**Logs**

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS with Version: Windows 1909
 - SDK Version:  1.3.0
 - Firmware version:  1.6.102

**Additional context**

wes-b, 2019-12-20:
* In Settings -> System -> Sound are you able to select the Azure Kinect Microphone and have it register sound with the 'Test your microphone' meter?
* Also check Settings -> Privacy -> Microphone and ensure it is still enabled.
* In Device Manager change View -> Devices by Container and confirm the "Azure Kinect 4K Camera"  container also has the Microphone Array in it.
[image]

rfilkov, 2019-12-26:
- In Settings -> System -> Sound - The Azure Kinect microphone array is there, but no sound is registered with 'Test your microphone'. When I click Troubleshoot, it says 'Audio services not responding'. See the attached screenshot.
- In Settings -> Privacy -> Microphone is enabled.
- In Device Manager, when I change View -> Devices by Container the 'Azure Kinect 4K Camera'-container looks the same as your screenshot.

[image]

wes-b, 2019-12-27:
Thank you, this is good info. Something more fundamental is wrong on your PC. Have you tried removing any third party audio software?

rfilkov, 2020-01-03:
I don't think I use 3rd party audio software packages. Do you mean any specific software?

Here is something interesting: I have a second Azure camera that doesn't have the microphone issue on the same machine. Both have the same firmware. Changing USB ports doesn't matter in this case. The 1st camera always has the microphone problem, while the 2nd one doesn't. 

I start thinking this is some kind of camera-specific issue, but it started right after I updated the Windows OS to v1909. Before that everything worked as expected. Is there any diagnostic tool for Azure Kinect cameras?

wes-b, 2020-01-07:
There was an upgrade bug for the microphone about a year ago and was fixed by Windows. Please make sure you have all your updates applied, which it sounds like you have done.

Next try to uninstall the audio driver. 
  1. With only the problematic Kinect attached, go to this device manager view https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1001#issuecomment-567989119
  1. Right click on the microphone array and select uninstall device
  1. Click uninstall. 
  1. Detach and reattach the Kinect and let the driver reinstall.

rfilkov, 2020-01-09:
Sorry, I did it twice but it did not help.

wes-b, 2020-01-09:
Can you confirm that mic works as expected on another PC?

rfilkov, 2020-01-13:
Yes, it works as expected on another machine.

wes-b, 2020-01-21:
We have confirmation there is a upgrade issue effecting the microphone. We are working on a fix.

closed 2020-03-05 by wes-b

wes-b, 2020-03-05:
Windows Update KB4516421 has been published. To get this fix ensure your PC is current on all its updates.

rfilkov, 2020-03-06:
Yes, after the latest Windows update the issue seems resolved. Thank you again!