microsoft/Azure-Kinect-Sensor-SDK #1003: Recorded video requires too much storage space

1216704928, 2019-12-20:
**Describe the bug**

I plan to record the mkv video first, store the video locally, and then input the video locally to track the human skeleton, but if I record the rgb, ir, and depth data streams, the saved mkv file is very large, which requires 4G in one minute. RAM. So I want to ask if there is any solution? I thought of a way to record only the rgb and depth data streams. Can human bone tracking only input the rgb and depth data streams, and not the ir data stream? I only input the rgb and depth data streams, the program will have errors, but because the k4abt_tracker_enqueue_capture () function is a dll format file, I ca n’t. I plan to record the mkv video first, store the video locally, and then save the video locally Input, track human bones, but if you record rgb, ir and depth data streams, the saved mkv file is very large and requires 4G of memory in one minute. So I want to ask if there is any solution? I thought of a way to record only the rgb and depth data streams. Can human bone tracking only input the rgb and depth data streams, and not the ir data stream? I only input the rgb and depth data streams, the program will have errors, but since the
 k4abt_tracker_enqueue_capture () function is from a dll format file, I cannot view the function.

**Expected behavior**

**Logs**

**Screenshots**

**Desktop (please complete the following information):**
 - OS with Version: 
 - SDK Version:  
 - Firmware version:  

**Additional context**

qm13, 2019-12-21:
Body tracking only requires the depth recording. You do not need to record the rgb camera (unless you have other reasons to record the rgb camera). We are investigating options to compress depth both at rest and in transit.

1216704928, 2019-12-21:
I only input a video that contains the depth and rgb data streams, but does not include the ir data stream, but an error occurs. If I only enter the depth data stream without the ir data stream, it seems that it is not possible to track human bones. An error occurs, why is it wrong?

[image]

qm13, 2019-12-21:
You need to input the depth and IR streams. You do not need the RGB stream.