microsoft/Azure-Kinect-Sensor-SDK #997: Version 0.9.5, small head turn and inaccurate eye, nose and ear positions

venscn, 2019-12-20:
**Describe the bug**
In 0.9.5, the head recognition is very inaccurate, hope to fix it in the next version

**Screenshots**
[image]
 - OS with Version: Windows 10
 - SDK Version:  0.9.4
 - Description: It's normal.

[image]
 - OS with Version: Windows 10
 - SDK Version:  0.9.5
 - Description: It's inaccurate.

prjova37, 2019-12-26:
Thank you for reporting this. We are investigating this issue.

Zumra, 2020-01-22:
@venscn Thank you for reporting the issue. We have addressed it, and the fix will be available in the next release. Looking forward to hearing your feedback

qm13, 2020-01-24:
Fixed in v1.0.0

closed 2020-01-24 by qm13