microsoft/Azure-Kinect-Sensor-SDK #998: Color Camera doesn't work

yeongwoonIm, 2019-12-20:
I launch the Azure Viewer only Color Camera.
but it doesn't work;;;(Depth Camera is fine. it works very well)

PC
OS : windows 10 enterprise 2016 LTSB 64bit
CPU : intel i7-7700K cpu @ 3.60GHz
RAM : 8 GB
GPU : Intel HD630

SDK : 1.3.0

Viewer shows this.
[image]

I searched issues. and I think this is similar problem.
[https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/839](url)

I checked Device Manager. it look like fine..
[image]

I did uninstall & install SDK and plug-in & plug-out USB cable several times. but it doesn't work..

what I have to do next??

wes-b, 2019-12-20:
You have something installed on your PC interfiering with your camera pipeline. Is anything in this https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/459#issuecomment-541994957 helpful?

tesych, 2020-01-06:
No response. Closing this issue. Please let us know if the problem continues.

closed 2020-01-06 by tesych

yeongwoonIm, 2020-01-10:
> You have something installed on your PC interfiering with your camera pipeline. Is anything in this [#459 (comment)](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/459#issuecomment-541994957) helpful?

There is no interfiering with my camera. it has only 1 kinectDK connected.