microsoft/Azure-Kinect-Sensor-SDK #999: onnx runtime error: [E:onnxruntime:K4ABT_ORT_Environment, cuda_call.cc:93 onnxruntime::CudaCall]

aistyler, 2019-12-20:
**Describe the bug**

I got an error when executing "body-tracking-samples/simple_sample"

**To Reproduce**

Build and Run  "body-tracking-samples/simple_sample"

**Expected behavior**

**Logs**

```txt
2019-12-20 17:27:48.8034913 [E:onnxruntime:K4ABT_ORT_Environment, cuda_call.cc:93 onnxruntime::CudaCall] CUDNN failure 4: CUDNN_STATUS_INTERNAL_ERROR ; GPU=0 ; hostname=X03000200040005 ; expr=cudnnFindConvolutionForwardAlgorithmEx( CudnnHandle(), s_.x_tensor, x_data, s_.filter_desc, w_data, s_.conv_desc, s_.y_tensor, y_data, 1, &algo_count, &perf, algo_search_workspace.get(), AlgoSearchWorkspaceSize);
2019-12-20 17:27:48.8040107 [E:onnxruntime:, sequential_executor.cc:165 onnxruntime::SequentialExecutor::Execute] Non-zero status code returned while running Conv node. Name:'Conv__2040' Status Message:
```

**Screenshots**

**Desktop (please complete the following information):**
 - OS with Version: Windows 1809
 - SDK Version:  1.3.0
 - Firmware version:  
```txt
Current Firmware Versions:
  RGB camera firmware:      1.6.102
  Depth camera firmware:    1.6.75
  Depth config file:        6109.7
  Audio firmware:           1.6.14
  Build Config:             Production
  Certificate Type:         Microsoft
```

**Additional context**

- Visual Studio version: Community 2017
- GPU: GeForce RTX 2060
```txt
C:\WINDOWS\system32>nvidia-smi
Fri Dec 20 17:44:39 2019
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 441.66       Driver Version: 441.66       CUDA Version: 10.2     |
|-------------------------------+----------------------+----------------------+
| GPU  Name            TCC/WDDM | Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  GeForce RTX 2060   WDDM  | 00000000:01:00.0 Off |                  N/A |
| 29%   29C    P8     8W / 190W |    487MiB /  6144MiB |      9%      Default |
+-------------------------------+----------------------+----------------------+
```
- nvidia-smi query output
```txt
C:\WINDOWS\system32>nvidia-smi --query

==============NVSMI LOG==============

Timestamp                           : Fri Dec 20 17:49:19 2019
Driver Version                      : 441.66
CUDA Version                        : 10.2

Attached GPUs                       : 1
GPU 00000000:01:00.0
    Product Name                    : GeForce RTX 2060
    Product Brand                   : GeForce
    Display Mode                    : Enabled
    Display Active                  : Disabled
    Persistence Mode                : N/A
    Accounting Mode                 : Disabled
    Accounting Mode Buffer Size     : 4000
    Driver Model
        Current                     : WDDM
        Pending                     : WDDM
    Serial Number                   : N/A
    GPU UUID                        : GPU-c26c64cb-0618-4e1e-d1d7-226b231fb85b
    Minor Number                    : N/A
    VBIOS Version                   : 90.06.3F.00.7F
    MultiGPU Board                  : No
    Board ID                        : 0x100
    GPU Part Number                 : N/A
    Inforom Version
        Image Version               : G001.0000.02.04
        OEM Object                  : 1.1
        ECC Object                  : N/A
        Power Management Object     : N/A
    GPU Operation Mode
        Current                     : N/A
        Pending                     : N/A
    GPU Virtualization Mode
        Virtualization Mode         : None
        Host VGPU Mode              : N/A
    IBMNPU
        Relaxed Ordering Mode       : N/A
    PCI
        Bus                         : 0x01
        Device                      : 0x00
        Domain                      : 0x0000
        Device Id                   : 0x1F0810DE
        Bus Id                      : 00000000:01:00.0
        Sub System Id               : 0x1F0810DE
        GPU Link Info
            PCIe Generation
                Max                 : 3
                Current             : 1
            Link Width
                Max                 : 16x
                Current             : 16x
        Bridge Chip
            Type                    : N/A
            Firmware                : N/A
        Replays Since Reset         : 0
        Replay Number Rollovers     : 0
        Tx Throughput               : 2280000 KB/s
        Rx Throughput               : 6000 KB/s
    Fan Speed                       : 29 %
    Performance State               : P8
    Clocks Throttle Reasons
        Idle                        : Active
        Applications Clocks Setting : Not Active
        SW Power Cap                : Not Active
        HW Slowdown                 : Not Active
            HW Thermal Slowdown     : Not Active
            HW Power Brake Slowdown : Not Active
        Sync Boost                  : Not Active
        SW Thermal Slowdown         : Not Active
        Display Clock Setting       : Not Active
    FB Memory Usage
        Total                       : 6144 MiB
        Used                        : 490 MiB
        Free                        : 5654 MiB
    BAR1 Memory Usage
        Total                       : 256 MiB
        Used                        : 229 MiB
        Free                        : 27 MiB
    Compute Mode                    : Default
    Utilization
        Gpu                         : 14 %
        Memory                      : 1 %
        Encoder                     : 0 %
        Decoder                     : 0 %
    Encoder Stats
        Active Sessions             : 0
        Average FPS                 : 0
        Average Latency             : 0
    FBC Stats
        Active Sessions             : 0
        Average FPS                 : 0
        Average Latency             : 0
    Ecc Mode
        Current                     : N/A
        Pending                     : N/A
    ECC Errors
        Volatile
            SRAM Correctable        : N/A
            SRAM Uncorrectable      : N/A
            DRAM Correctable        : N/A
            DRAM Uncorrectable      : N/A
        Aggregate
            SRAM Correctable        : N/A
            SRAM Uncorrectable      : N/A
            DRAM Correctable        : N/A
            DRAM Uncorrectable      : N/A
    Retired Pages
        Single Bit ECC              : N/A
        Double Bit ECC              : N/A
        Pending Page Blacklist      : N/A
    Temperature
        GPU Current Temp            : 29 C
        GPU Shutdown Temp           : 93 C
        GPU Slowdown Temp           : 90 C
        GPU Max Operating Temp      : 88 C
        Memory Current Temp         : N/A
        Memory Max Operating Temp   : N/A
    Power Readings
        Power Management            : Supported
        Power Draw                  : 14.68 W
        Power Limit                 : 190.00 W
        Default Power Limit         : 190.00 W
        Enforced Power Limit        : 190.00 W
        Min Power Limit             : 125.00 W
        Max Power Limit             : 225.00 W
    Clocks
        Graphics                    : 690 MHz
        SM                          : 690 MHz
        Memory                      : 810 MHz
        Video                       : 630 MHz
    Applications Clocks
        Graphics                    : N/A
        Memory                      : N/A
    Default Applications Clocks
        Graphics                    : N/A
        Memory                      : N/A
    Max Clocks
        Graphics                    : 2160 MHz
        SM                          : 2160 MHz
        Memory                      : 7001 MHz
        Video                       : 1950 MHz
    Max Customer Boost Clocks
        Graphics                    : N/A
    Clock Policy
        Auto Boost                  : N/A
        Auto Boost Default          : N/A
    Processes
...{OMIT}...
```

yijiew, 2019-12-20:
@aistyler Thanks for reporting the issue. Does the Azure Kinect Body Tracking Viewer that you installed with the msi installer works for you?

aistyler, 2019-12-22:
@yijiew No. The Azure Kinect Body Tracking Viewer also does not work with the same error.

hiptkang, 2020-01-05:
@aistyler Do you have any solution for this error?

aistyler, 2020-01-06:
@yijiew Hi, Is there something new?

yijiew, 2020-01-06:
@aistyler Sorry for the late response. I took some vacation for the holiday season. Thanks for trying the official tool comes with MSI. It might be a ONNX runtime issue. I will engage with their team to ask for their suggestions.

aistyler, 2020-01-07:
@yijiew Ok. Thanks for your support.

yijiew, 2020-01-13:
Hi @aistyler , we tried and we couldn't repro this issue on our machine. There are two main differences between your settings and our machines. 
1. The body tracking SDK requires cuda 10.0. And I realized that you were using CUDA 10.2. If it is possible, could you uninstall CUDA 10.2 and install 10.0 to see whether this problem could be resolved? 
2. Another difference is that the Graphics card. We have tested on various versions of RTX and GTX cards (Like GTX 1050Ti, 1060, 1070, 1080, RTX 2070, RTX 2080), but we don't have a RTX 2060 at hand. It would be a very helpful information if you happened to have other graphics card to try it out and confirm whether it is an issue related to graphics card itself. 

Another test you can run that might provide us with more information is running the model through CPU. We now supported CPU mode from our Azure Kinect Body Tracking Viewer. It can be run by the following steps: 
1. go to the "Azure Kinect Body Tracking SDK" installed folder\tools
2. run "k4abt_simple_3d_viewer.exe CPU".
3. you should be able to see the body tracking SDK running on CPU. 
Could you help to test out whether at least the CPU mode runs correctly on our machine? 

Thanks!

closed 2020-01-24 by qm13