microsoft/Azure-Kinect-Sensor-SDK #1005: Driver issues with Kinect Azure

DivyanshK86, 2019-12-21:
- My Azure kinect is connected to USB 3.0 port.
- The light at the back is stable white.
- Windows camera app works fine.

I am having these driver issues:
 
[image]

I have installed the Azure kinect viewer v1.3.0 but it does not find the device:

[image]

wes-b, 2019-12-21:
In device manager, what USB host controller are you connected to?

jelmerS2, 2020-01-02:
Strange, at my installation the Azure Kinect is installed succesfully and it shows up in the Azure Kinect Viewer. 

But when I try to start the videofeed, the program quits without warning. 
The same behaviour with another program that uses the new Kinect. Exit without any error.

wes-b, 2020-01-07:
Are there any issues reported in Event Viewer https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/853#issuecomment-567119412?

If not can you repro the issue with k4aviewer and share the log?

Chris45215, 2020-01-07:
I've seen this happen when the camera is connected to the wrong USB port, or one that isn't a compatible host controller. The cameras are very picky about the port type they are connected to - and some computers have a USB 3.1 or USB 3.2 port which actually isn't compatible. ASRock's ITX Ryzen motherboards, for example, include a special extra USB 3.1 port (separate from all the USB 3.0 and 2.0 ports), and that port has its own dedicated host controller - but it is not compatible with the camera. All the other USB 3.0 ports on the motherboard are compatible with the camera, so it's just a matter of attaching to the correct port.

wes-b, 2020-01-14:
@jelmerS2 Let us know if you are still working through this.

closed 2020-01-14 by wes-b