microsoft/Azure-Kinect-Sensor-SDK #1007: Body Tracking SDK 0.9.5  - Ubuntu - CPU Usage near 100%

RoseFlunder, 2019-12-22:
**Describe the bug**
With the current Sensor SDK 1.3.0 and Body tracking SDK 0.9.5. the CPU usage climbs to 100%.
The CPU is an six core Intel i5-9600KF @ 4,6GHz  so way above the minimum specs.
Because of this the CPU fan goes full speed and the PC becomes hot and loud after some minutes.
The GPU Usage of the NVidia 2070 Super is around 50%, so thats looks fine to me.
Is this high CPU usage the expected behaviour?

**To Reproduce**
1. Start the k4abt_simple_3d_viewer
2. Observe the CPU usage with a tool like top or htop

**Expected behavior**
Less CPU usage.

**Desktop (please complete the following information):**
 - OS with Version: Ubuntu 18.04
 - SDK Version:  1.3.0 & 0.9.5
 - Firmware version:  Current firmware which is delivered with 1.3.0
- CPU: Intel i5-9600KF
- GPU: NVidia 2070 Super
- RAM: 16GB

yijiew, 2020-01-06:
Hi @RoseFlunder , ONNX runtime leverage OpenMP to optimize its performance. The 100% CPU usage might be caused by the OpenMP settings in your machine. Could you try to set the OMP_WAIT_POLICY to Passive to see whether it solves your issue? 
FYI: https://www.smcm.iqfr.csic.es/docs/intel/compiler_f/main_for/optaps/common/optaps_par_var.htm

RoseFlunder, 2020-01-07:
@yijiew Thank you for your help. I can confirm that your suggestion worked for me.

closed 2020-01-07 by RoseFlunder

yijiew, 2020-01-09:
@RoseFlunder Thanks for confirming the mitigation works. We will keep investigating why the new SDK version use OMP_WAIT_POLICY "Active" by default.