microsoft/Azure-Kinect-Sensor-SDK #1009: Timeout bug in CMFCameraReader

rfilkov, 2019-12-26:
**Describe the bug**

I think I found a bug in CMFCameraReader::Stop()-method. There is a do..while(true) loop at the end of this method that waits for 'm_hStreamFlushed' or times out after 10 seconds. The problem is, if the waiting times out, it continues to wait and time out again, instead of just return from the method. This may cause thread hanging, when the color camera is enabled and sensor gets disconnected. I think replacing 'break' with 'return' in 'case WAIT_TIMEOUT:' would fix the bug.

**To Reproduce**

1. Build 'Azure-Kinect-Sensor-SDK' from source.
2. Start 'viewer_opengl.exe'-example with logging to file enabled.
3. Disconnect the sensor, without stopping the app.
4. Wait for a minute or two, without stopping the app.
5. Look at the log-file. There should be multiple 'Stop(). Timeout waiting for m_hStreamFlushed'-messages at the end, instead of only one.

**Expected behavior**

The method should return after the 1st timeout. There should be only one 'Stop(). Timeout waiting for m_hStreamFlushed' per log session.

**Logs**

Attached.

[k4a.log](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/files/4002359/k4a.log)

**Screenshots**

**Desktop (please complete the following information):**
 - OS with Version: 
 - SDK Version:  
 - Firmware version:  

**Additional context**

wes-b, 2019-12-27:
Good find!

rfilkov, 2020-01-03:
This is in mfcamerareader.cpp, line 531.

closed 2020-01-09 by wes-b

rfilkov, 2020-01-22:
The issue is resolved. Thank you again!
By the way, is there a way to contact you privately? I'd like to send something as a little form of appreciation.

wes-b, 2020-01-24:
Your thanks is enough, I am happy to help :). My email address is available as part of my commit history if you want to chat. ```git log``` from your local clone will help you with that.

PR #1024 (2020-01-09): Moving wait loop to only happen when Flush() succeeds
## Fixes #1009 

### Description of the changes:
- This wait loop has been moved such that it is only called when Flush() has succeeded and we know the function was called.