microsoft/Azure-Kinect-Sensor-SDK #1012: Why there is no transformation from PassiveIR Image to Color Camera?

Piler-Tam, 2019-12-30:
Similar the link below, depth image originally is from a fisheye camera but it can change to normal view by the transformation functions. However, there is no transformation function for IR images. So how can I do transformation for IR images? Thanks.
https://docs.microsoft.com/en-us/azure/kinect-dk/use-image-transformation

rabbitdaxi, 2019-12-30:
@Piler-Tam Thanks for your feedback. 
- If you are talking about active IR images for those depth modes (NFOV or WFOV with binned or unbinned), the depth camera produce both depth image and ir image from a single frame, they share the same camera intrinsics and extrinsics. If you want to transform ir image, you can look at this API: [k4a_transformation_depth_image_to_color_camera_custom()](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/group___functions_gac00dd00e7612a86382e3d0a130f276bb.html#gac00dd00e7612a86382e3d0a130f276bb) and you can use the custom_image parameter to transform the payload from depth camera space to color space. Here, you can pass the IR image to the custom_image as  K4A_IMAGE_FORMAT_CUSTOM16 format and then select the interpolation type that you want.
- If you are talking about passive IR image only, it is not possible to transform if you do not know the depth image of the scene. Please remember, passive IR means there is no active IR illumination from the depth camera, therefore, there is no depth in this mode. However, if you are setting your camera static and scene static, you can try to capture a depth image using WFOV unbinned mode, and then switch to passive IR mode to capture a passive IR image, and then use custom image transform API above to transform your passive IR image with the depth image for the exactly same scene.

Piler-Tam, 2019-12-31:
@rabbitdaxi Thanks for your reply. 

I am using C# but there seems no corresponding API that you mention in C. How can I do it in C#?

rabbitdaxi, 2019-12-31:
@Piler-Tam Thanks for pointing out. I see that there is another issue #617 which is tracking the C# API for this functionality of transform custom image, which is triage approved but not done yet. 

And yes, if you are talking about passive IR, since you do need a depth image for this transformation to work, you can try to use the idea of capture a WFOV_UNBINNED depth image and then followed by capture a Passive IR image and assume during the capture of these two frames the scene is static.

Piler-Tam, 2020-01-02:
@rabbitdaxi Thanks.

Since the C# API is missing, is that mean there is no way to transform passive IR image in C# now?

rabbitdaxi, 2020-01-06:
@Piler-Tam some thoughts on your questions:
- Yes, the C# API for the transforming custom image is not there, will notify the team who works on the C# API to prioritize it. If you can work with C++/C, this API is available. But, yes, the C# API for this function need to be added. 
- The operating range is meant for depth, since the Passive IR is not a depth mode, therefore, it was labeled as N/A, however, you can get Passive IR signal in the image as long as you have strong enough external IR source signal with certain max distance depends on the external IR source. The native sensor resolution will sense all the external IR signal depends on how you illuminate the scene. The native sensor resolution is 1024x1024 and close to 120x120 degree same as WFOV. 
- For your goal, if the distortion of the image is the issue, you can try to undistort the image using the camera intrinsics. We have an example in the github for undistortion. If you want to detection some point and transform to color camera for that single point/pixel, you can call k4a_calibration_2d_to_2d or it correspondent C# API.

closed 2020-01-21 by rabbitdaxi