microsoft/Azure-Kinect-Sensor-SDK #1013: one PC, six Kinect sensor Multi thread programming

devLupin, 2019-12-30:
I want to control multiple sensors from one computer.

The idea was to use threads. In the code below I tried to do multithreaded programming using std :: thread and std::mutex.
```
uint32_t device_count = k4a_device_get_installed_count();
for (uint8_t deviceIndex = 0; deviceIndex < device_count; deviceIndex++){
		k4a_device_t temp_handle = NULL;
		if (K4A_RESULT_SUCCEEDED != k4a_device_open(deviceIndex, &temp_handle))
		{
			printf("%d: Failed to open device\n", deviceIndex);
			while (1);
		}

		k4a_device_configuration_t device_config = K4A_DEVICE_CONFIG_INIT_DISABLE_ALL;
		device_config.depth_mode = K4A_DEPTH_MODE_NFOV_UNBINNED;

		// Start the camera with the given configuration
		k4a::device device = k4a::device(temp_handle);
		device.start_cameras(&device_config);

		k4a::calibration sensor_calibration = device.get_calibration(device_config.depth_mode, device_config.color_resolution);

		k4abt::tracker tracker = k4abt::tracker::create(sensor_calibration);
}
...........................
```
But it doesn't seem to be possible to have more than one camera on at the same time. 
If one camera is opened by k4a_device_open (), no other camera can try anything.

After When k4a_device_open() and k4a_device_close() is called on camera 1  is called until k4a_device_open() on camera 2, the delay is too long.

I'm trying to multi-thread programming six sensors on one computer.

Reduce the time between k4a_device_close() and k4a_device_open(), or please tell me if you have other way.

Brent-A, 2020-01-06:
@devLupin, you should be able to access multiple sensors from separate threads. You can only have one instance of a sensor opened at a time, so you need to ensure that if one thread has opened a device another thread does not attempt to open the same device index.

In the code example you posted you didn't show how you were using threading, do you have a more complete example you can point us to?

You may also be able to get more information on why there is a failure by [enabling logging](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/group___logging.html).

devLupin, 2020-01-07:
@Brent-A 

Refer to Microsoft/Azure-Kinect-Samples/simple_cpp_sample

Below is my main() code.

```cpp
int main(void)
{
        // MyJoint is defined class
	MyJoint* info = new MyJoint[6];

	uint32_t device_count = k4a_device_get_installed_count();
	std::cout << "num of connected k4a : " << device_count << '\n' << '\n';
	if (device_count == 0)
	{
		printf("No k4a devices attached!\n");
		while (1);
	}

	/* Need extra cording work */
	std::string member_RFID = "_159";
	std::mutex m;
	
	std::vector<std::thread> workers;
	for (uint8_t i = 0; i < device_count; i++) {
		workers.push_back(std::thread{ &MyJoint::DetectJoint, &info[i], i, member_RFID, std::ref(m) });
	}

	for (uint8_t deviceIndex = 0; deviceIndex < device_count; deviceIndex++) {
		workers[deviceIndex].join();
	}
}
```

Below is DetectJoint() Func

```cpp
void MyJoint::DetectJoint(int deviceIndex, std::string rfid, std::mutex& m)
{
	try
	{
		k4a_device_t temp_handle = NULL;
		if (K4A_RESULT_SUCCEEDED != k4a_device_open(deviceIndex, &temp_handle))
		{
			printf("%d: Failed to open device\n", deviceIndex);
			while (1);
		}

		k4a_device_configuration_t device_config = K4A_DEVICE_CONFIG_INIT_DISABLE_ALL;
		device_config.depth_mode = K4A_DEPTH_MODE_NFOV_UNBINNED;

		// Start the camera with the given configuration
		k4a::device device = k4a::device(temp_handle);
		device.start_cameras(&device_config);

		k4a::calibration sensor_calibration = device.get_calibration(device_config.depth_mode, device_config.color_resolution);

		k4abt::tracker tracker = k4abt::tracker::create(sensor_calibration);

		int frame_count = 0;
		while (frame_count < LOOP_CNT)
		{
			k4a::capture sensor_capture;
			if (device.get_capture(&sensor_capture, std::chrono::milliseconds(K4A_WAIT_INFINITE)))
			{
				frame_count++;
				std::cout << "Start processing frame " << frame_count << '\n';

				if (!tracker.enqueue_capture(sensor_capture))
				{
					// It should never hit timeout when K4A_WAIT_INFINITE is set.
					std::cout << "Error! Add capture to tracker process queue timeout!" << '\n';
					while (1);
				}

				k4abt::frame body_frame = tracker.pop_result();
				if (body_frame != nullptr)
				{
					
					uint32_t num_bodies = body_frame.get_num_bodies();
					std::cout << num_bodies << " bodies are detected!" << '\n';

					for (uint32_t i = 0; i < num_bodies; i++)
					{
						k4abt_body_t body = body_frame.get_body(i);
						body_information(body, frame_count, rfid);
					}

					
				}
				else
				{
					//  It should never hit timeout when K4A_WAIT_INFINITE is set.
					std::cout << "Error! Pop body frame result time out!" << '\n';
					while (1);
				}
			}
			else
			{
				// It should never hit time out when K4A_WAIT_INFINITE is set.
				std::cout << "Error! Get depth frame time out!" << '\n';
				while (1);
			}
		}
		// std::cout << "Finished body tracking processing!" << '\n';
	}
	catch (const std::exception & e)
	{
		std::cerr << "Failed with exception:" << '\n'
			<< "    " << e.what() << '\n';
		while (1);
	}
}
```

When I run these codes, I get the following error

[image]

So, using lock() and unlock()

```cpp
void MyJoint::DetectJoint(int deviceIndex, std::string rfid, std::mutex& m){
        m.lock();
        try{........}
        catch{........}
        m.unlock();
}
```

I think, At the same time, however, the camera did not appear to be controlled.
k4a_device_open([1st camera], & temp_handle);
k4a_device_open([2nd camera], & temp_handle);
..................
k4a_device_open([6th camera], & temp_handle);

Is there a way to control the camera at the same time for a certain period of time?
In kinect 6 we need to use the data value of k4a_float_t.

Thank you.

devLupin, 2020-01-07:
@Brent-A 

I soved this problem through k4a_device_t handle[6].
But I tested using two azure kinect devices, so six kinect running is not guaranteed.

When running code, two kinect device red led lights up at the same time. I think it is two kinect devices are opened.

Will this be a problem in the future?

The code below is two kinect multi threading code.
The difference from the previous code is that no mutexes are used.

```cpp
uint32_t device_count = k4a_device_get_installed_count();
	std::cout << "num of connected k4a : " << device_count << '\n' << '\n';
	if (device_count == 0)
	{
		printf("No k4a devices attached!\n");
		while (1);
	}

k4a_device_t temp_handle[6] = { NULL };
	for (uint8_t i = 0; i < device_count; i++) {
		if (K4A_RESULT_SUCCEEDED != k4a_device_open(i, &(temp_handle[i])))
		{
			printf("%d: Failed to open device\n", i);
			while (1);
		}

		workers.push_back(std::thread{ &MyJoint::DetectJoint, &info[i], i, member_RFID,  temp_handle[i] });
	}

	for (uint8_t deviceIndex = 0; deviceIndex < device_count; deviceIndex++) {
		try { 
			workers[deviceIndex].join(); 
		} catch (std::exception ex) {
			printf("join() error log : %s\n", ex.what());
			while (1);
		}
	}
```

Brent-A, 2020-01-07:
I see. It looks like the problem isn't with having multiple devices open at the same time, but it may be with the internal library getting initialized simultaneously at the same time.

In your new code, you open the devices in series, and then use them in parallel, which avoids parallel problems during SDK initialization.

This looks like issue #276 

@devLupin, it looks like you might be using v1.2 or earlier of the SDK, can you please try moving to v1.3?

wes-b, 2020-01-07:
Should be linked to issues #676 & #769. Edited https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1013#issuecomment-571638663 to point at correct issue.

devLupin, 2020-01-08:
@Brent-A @wes-b 
Thank you for your answer.

I develop using 1.3ver but i don't know body tracking sdk version.
I read issue #676 and #769 

I will test and ask again. Thanks

Brent-A, 2020-01-14:
Thanks @devLupin. I'll close this issue. Let us know if there is further trouble.

closed 2020-01-14 by Brent-A