microsoft/Azure-Kinect-Sensor-SDK #1015: Camera recognised but black and timeout

jelmerS2, 2020-01-03:
**Describe the bug**
I have problems with all SDK related programs, they quit when trying to access the Azure Kinect.
Logging doesn't work, I get 
ERROR: Logger unable to open log file "logs/k4a.log"...

The only hint so far: in the windows camera app, the Kinect is recognized, but the image remains black. 
When pressing the shutter button, I get PhotoCaptureTimeOut.

PS: why can't I upload images here?

tesych, 2020-01-06:
@jelmerS2 , thank you for your question. Please check out how to collect the logs in our [Azure Kinect documentation](https://docs.microsoft.com/en-us/azure/Kinect-dk/troubleshooting#collecting-logs).

closed 2020-01-08 by tesych

wes-b, 2020-01-08:
You probably didn't have a folder called 'logs' from the location you executed. 

Look at some of our issues about USB controllers and make sure you are using a supported host controller.