microsoft/Azure-Kinect-Sensor-SDK #1017: Skeleton need smoothing and set_temporal_smoothing need introduction

wqs111000, 2020-01-06:
By default,  the skeleton  is easier to shake. I find the function set_temporal_smoothing but its introduction is plain.  I've done some experiments, but I'm not sure.
How to smooth skeleton to minimize the shaking? How does the value of smoothing_factor influence the results? Its range？
```
 /* Set temporal smoothing */
 void set_temporal_smoothing(float smoothing_factor) noexcept
{
     k4abt_tracker_set_temporal_smoothing(m_handle, smoothing_factor);
}
```

yijiew, 2020-01-06:
@wqs111000 It seems that you are using the CPP wrapper for your header. For detailed documentation, please refer to the original C API descriptions: 

 /** Control the temporal smoothing across frames.
 *
 * \param tracker_handle
 * Handle obtained by k4abt_tracker_create().
 *
 * \param smoothing_factor
 * Set between 0 for no smoothing and 1 for full smoothing. Less smoothing will increase the responsiveness of the
 * detected skeletons but will cause more positional and orientational jitters.
 *
 * \relates k4abt_tracker_t
 *
 * \remarks
 * The default smoothness value is defined as K4ABT_DEFAULT_TRACKER_SMOOTHING_FACTOR.
 */
 K4ABT_EXPORT void k4abt_tracker_set_temporal_smoothing(k4abt_tracker_t tracker_handle, float smoothing_factor);

closed 2020-01-06 by yijiew

VegetableWithChicken, 2021-07-12:
I also encountered this problem. Have you solved it