microsoft/Azure-Kinect-Sensor-SDK #1018: Using Body Tracking functions won't compile

benjamminhansen, 2020-01-06:
**Describe the bug**

When trying to use the Body Tracking functions, the code gives a LINK2019 error such as:

`unresolved external symbol _k4abt_tracker_create reference in` my function.

It says that on any functions I use like: `k4abt_tracker_destroy()` and `k4abt_tracker_pop_result()` and any Body Tracking function I try to use.

I know it's linker error and not any other because it only occurs when the object files are trying to be combined. Intellisense doesn't catch any errors, and I can use the Body Tracking types like `k4abt_tracker_t` and `k4abt_tracker_configuration_t`.

**To Reproduce**

I expected the project to build successfully.

**Logs**

**Screenshots**

**Desktop (please complete the following information):**
 - OS with Version: Windows 10
 - SDK Version:  0.9.5
 - Firmware version:  Not Applicable

**Additional context**
I had tried the body tracking functions earlier and Intellisense notified me that it couldn't find the function because I didn't have the Body Tracking SDK even installed... So when I followed the instructions on the docs for installing the SDK, I thought it would work. Then, I got the linker errors.

yijiew, 2020-01-06:
@benjamminhansen How did you include the Body Tracking SDK to your project? Did you add the libs to your lib path? 
The suggested way to bring in the Body Tracking SDK is by Nuget package. If you just want a sample that is already working. You can download the visual studio projects from here: https://github.com/microsoft/Azure-Kinect-Samples/tree/master/body-tracking-samples

benjamminhansen, 2020-01-06:
@yijiew Thank you for the ideas. I have installed the Body Tracking SDK through the Nuget package. I have basically the same configuration file as the simple-sample from the GitHub samples you linked me to. 
Here is packages.config:
[image]

For some reason, Intellisense is saying this about the `<packages>`:
[image]

Not sure if that matters. If I use Nuget, do I need to do anything else to "hook" everything up? I'm not sure how to add the libs to my lib path.

benjamminhansen, 2020-01-06:
I've tried to compile the int main() function given in the simple-sample and the Body Tracking functions still won't compile.

yijiew, 2020-01-07:
I realized that you are using the Sensor SDK 1.4.0-alpha. The current official Body Tracking SDK depends on Sensor SDK 1.3.0. Could change the Sensor SDK nuget package version to give it a try again?

benjamminhansen, 2020-01-07:
I will try that.

benjamminhansen, 2020-01-07:
I tried that, using NuGet and the error is still there.

benjamminhansen, 2020-01-07:
@yijiew Thanks for your sincere help. If you didn't notice, this is the same problem as [https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/750](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/750). I followed the comments in that thread and solved the issue. My platform for debugging and building was Win32, not x64. That fixed my issue.

Before posting I searched previous issues looking for similar titles or phrases, but didn't find it until today searching every issue ever posted under my search: "body tracking". Maybe changing or adding something to the title of [https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/750](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/750) would help developers in the future solve this issue.

Thanks!

closed 2020-01-07 by benjamminhansen

yijiew, 2020-01-09:
@benjamminhansen : Thanks for confirming the root cause. There were already three people asking about this issue. Do you think adding some notes on the root page of the sample folder will help? 
https://github.com/microsoft/Azure-Kinect-Samples/tree/master/body-tracking-samples