microsoft/Azure-Kinect-Sensor-SDK #1021: Does Azure Kinect have recording capability now

1216704928, 2020-01-07:
Does Azure Kinect have recording capability now?Hopefully the videos recorded by Azure Kinect will have sound.

**Is your feature request related to a problem? Please describe.**

**Describe the solution you'd like**

**Describe alternatives you've considered**

**Additional context**

tesych, 2020-01-07:
@1216704928 thank you for your question! Azure Kinect Sensor SDK includes video recording capabilities. To include audio, please submit your idea to our [customer voice]{https://feedback.azure.com/forums/920053}, so other users can vote for your idea.

closed 2020-01-07 by tesych

rfilkov, 2020-01-09:
There is already a similar idea there: https://feedback.azure.com/forums/920053-azure-kinect-dk/suggestions/38424175-no-microphone-access-available-in-api-recorder-k4 

Please upvote it and share it to other interested users, as well.