microsoft/Azure-Kinect-Sensor-SDK #1025: kinfu crash when cover sensor

timhutton, 2020-01-10:
Using latest code on Azure-Kinect-Samples, with OpenCV 4.1.2. (Note that there is no issue tracker on that repo; it says to use this one.)

To reproduce:
- start the opencv-kinfu sample
- cover the sensor

The sample stops and prints the error:
```
[ERROR:0] global H:\OpenCV-4.1.2\sources\modules\core\src\ocl.cpp (3039)
cv::ocl::Kernel::set OpenCL: Kernel(fillPtsNrm)::set(arg_index=14, flags=260):
can't create cl_mem handle for passed UMat buffer (addr=0000004D17DAE480)
```

This is unexpected behavior. Covering the sensor to reset the tracking is an established action, as seen in demos of KinectFusion elsewhere.

(Bug may be in opencv code upstream but needs triage here to identify the cause.)

rabbitdaxi, 2020-01-10:
@timhutton hi tim, the sample code is simply using the opencv kinfu module, and the crash you mentioned is inside of opencv kinfu. You may better open this issue in opencv contrib github repo. https://github.com/opencv/opencv_contrib/issues

timhutton, 2020-01-10:
OK, thanks. Created: https://github.com/opencv/opencv_contrib/issues/2404

closed 2020-01-10 by timhutton