microsoft/Azure-Kinect-Sensor-SDK #1027: Bad camera feed

nicobeek, 2020-01-14:
**Describe the bug**
The video feed sometimes looks like it is missing some h-sync pulses, causing some frames to look like a sliding puzzle. The image looks similar to a bad satellite feed in a thunderstorm.

**To Reproduce**
I have not been able to reproduce the error, but I have seen it happen several times.
Restarting the program, or using a different program does not fix the issue. The only way to fix the issue is unplugging the USB cable. I have a recorded video clip of a bad video feed available.

**Desktop (please complete the following information):**
 - OS with Version: Microsoft Windows 10 Pro Version 10.0.18362 Build 18362
 - SDK Version:  1.3.0
 - Firmware version:  
  RGB camera firmware:      1.6.102
  Depth camera firmware:    1.6.75
  Depth config file:        6109.7
  Audio firmware:           1.6.14

wes-b, 2020-01-14:
We have not heard of this issue before and it sounds like it might be hardware related. Do you have another Kinect you could use for testing? Or another PC? How about a different camera?

Also seeing the video would be helpful if you can share it.

nicobeek, 2020-01-14:
[MainWindow 2020-01-10 08-17-18.zip](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/files/4061800/MainWindow.2020-01-10.08-17-18.zip)

Really hard to rule out stuff when the problem only shows up intermittently. I can try using a different Kinect/PC, but that doesn't proof much, since the setup I'm currently using has been working with no issues for the past few days too. Anyways, maybe the attached video adds some value

wes-b, 2020-01-24:
Closing this for now as we don't have any other reports of this. We can investigate more if we see more instances.

closed 2020-01-24 by wes-b