microsoft/Azure-Kinect-Sensor-SDK #1032: how to reduce depth noise at human hair part ?

carlhuang070, 2020-01-16:
And now , we found every depth camera may be have noise issue at human hair part.

So we want to know whether Kinect DK SDK have api which can increase the power of IR emiting to reduce the hair noise ?

tesych, 2020-01-16:
@carlhuang070 , can you share more details about what you see including the details about human hair (color for example)?

rabbitdaxi, 2020-01-16:
@carlhuang070 may be try use NFOV modes see whether it helps (or try to have the camera move closer to the hair). The noise is a function of signal, therefore, for some material which is IR absorbing could reduce the signal returned back dramatically and further increasing the noise.

closed 2020-01-23 by tesych