microsoft/Azure-Kinect-Sensor-SDK #1035: Azure Connect color camera does not work in Azure Kinect Viewer

tyleringebrand, 2020-01-16:
**Describe the bug**
I've just begin using the Azure Kinect and I cannot get the color camera to work in the Azure Kinect Viewer program. When I hit start using the default settings, the screens stay black for a few seconds before it times out. I can disable color camera, and the other two cameras work fine(Infrared and Depth). I also found out I can view the color camera through Window's Camera App, which tells me the hardware is fine and it is connected properly. Fearing the Windows app or others was blocking the Kinect app from accessing it, I disabled every other App that uses a camera hoping that would allow it to access the camera feed. I've also ensured the firmware is correct, and I reduced the frame rate to reduce resources to see if that was an issue.

**To Reproduce**
Plugin Kinect, download software as in the instructions. 
Open Kinect Viewer. 
Click the device and click start (using default settings).

**Expected behavior**
All three cameras display when using the Kinect Viewer.

**Logs**
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:     16088 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:     49422 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:     82755 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    116088 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    149422 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    182755 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    216088 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    249422 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    282755 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    316088 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    349422 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    382755 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    416088 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    449422 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    482755 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    516088 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    549422 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    582755 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    616088 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    649422 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    682755 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    716088 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    749422 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    782755 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    816088 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    849422 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    882755 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    916088 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    949422 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    982755 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1016088 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1049422 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample

**Desktop (please complete the following information):**
 - OS with Version: Windows 10 Enterprise version 1607
 - SDK Version:  1.3.0

 Current Firmware Versions:
  RGB camera firmware:      1.6.102
  Depth camera firmware:    1.6.75
  Depth config file:        6109.7
  Audio firmware:           1.6.14
  Build Config:             Production
  Certificate Type:         Microsoft

**Additional context**
Thank you in advance. Any help would be greatly appreciated.

wes-b, 2020-01-17:
@tyleringebrand we have had this issue reported a couple times, #998, #839, #459. Unfortunately we have not been able to root cause it and we don’t have a way to reproduce it. We believe this issue is related to the configuration on your PC rather than the SDK itself.

  - Please capture full logs for this https://docs.microsoft.com/en-us/azure/Kinect-dk/troubleshooting#collecting-logs?
  - Does the Azure Kinect DK work with Windows camera app?
  - Does the Windows camera app work with the installed camera on the PC (if it has one)?
  - Does this same issue reproduce on another PC?
  - If you have a 2nd Azure Kinect, does this issue reproduce for that camera?
  - Would you consider capturing a Media Foundation trace and sharing it?
    - ```mftrace -l 4 -es -o trace.txt bin\k4aviewer.exe```
    - stream the color sensor for 5s and close.
    - https://docs.microsoft.com/en-us/windows/win32/medfound/using-mftrace

tyleringebrand, 2020-01-17:
1. I captured the logs and attached them in a text file

2. The Azure color camera did work with the camera app while the problem persisted, however I then deleted the Windows Camera app thinking that it might be blocking the Kinect app from accessing the camera pipeline. 

3. As I deleted the app and the Microsoft store is also having problems, I can't check today. However as it worked for the Azure and this computer has been used for camera application development before, I assume it worked for other cameras as well. 

4. I tried the Kinect camera on another computer, and it worked properly. I'd still like to resolve this issue however. 

5. I do not have another Kinect camera to try it with

6. Could you be a bit more specific on how to do this, I am new to Windows development.

tyleringebrand, 2020-01-17:
Log is at https://gist.github.com/tyleringebrand/3b91819ac6d71d27ce6acbdb3dc7b09c

wes-b, 2020-01-17:
Summarizing:
1. Thanks for the log, that didn't show anything sadly
2. Windows Camera app works with Azure Kinect DK camera
3. Unknown if another camera works
4. The Camera Works Perfectly on another PC
5. 2nd Azure Kinect not available
6. More instruction needed. - Can you be specific about the help you need. I shared the command to run and the link to more info. You need to install a Windows SDK to get the tool, for example this might work. The install is long and I have not tried it. https://developer.microsoft.com/en-US/windows/downloads/windows-10-sdk

Because the camera works fine on another PC, I suggest going into add/remove programs and remove all video/camera related softwear, even remove audio related software as I have seen that have a negative impact on the Video pipeline.

wes-b, 2020-01-21:
@tyleringebrand any update?

tyleringebrand, 2020-01-21:
Deleted 13 different audio/video related programs, no luck. Also tried looking through windows Services for things out of place, no luck. I am now reinstalling Windows as a last resort.

tyleringebrand, 2020-01-23:
Reinstalling windows successfully fixes the problem. For those visiting this page later, reinstalling is likely quicker than finding the program that is causing issues.

closed 2020-01-23 by tyleringebrand

wes-b, 2020-01-23:
Thanks for the update, please let us know if the issue shows back up and what caused it.

LuciaVisuals, 2020-02-10:
@wes-b I am having this same issue with my Azure Kinect but I was not having it prior with the same PC. I AM using a USB 3.0 extender as I am connecting over a long distance, which I was NOT using prior. A couple questions:

1- Is it possible this issue comes from using extra cables?

2- If yes, how would you recommend best extending the camera cables to connect to a PC tower about 50 feet away (in cable length)? I have a USB 3.0 active extender with its own power adapter and signal booster.

3- I am using the camera through another software (TouchDesigner) which does not give me an option to disable the color camera so therefore it just doesn't work at all.

For context, I am using the Kinect for an interactive art piece and the show opens on Wednesday, 2/12/20. Everything worked fine at home (with shorter cables) but in the gallery it is not. 

I can share my logs but I'll have to do that tomorrow AM once I can grab them from the PC. The error message I am getting is "releasing capture due to full queue TS". When I disable the color camera in the Azure Kinect Viewer, I get depth and IR images with no lag or issues. 

Thank you!

wes-b, 2020-02-10:
Have you seen issue https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/427 about extension cables? 

Feel free to open a new issue (using the bug template) with your logs and we can see if it is the same or not

LuciaVisuals, 2020-02-10:
I just took a look. None of the cables suggested would solve my issue of
trying to extend 50 feet unfortunately. I am happy to report back
though that I tested all of my PC tower's USB 3.0 ports this morning with
the below extender and the Azure Kinect, and (oddly) the port on the front
of the machine works perfectly and solves the errors- the rest of them do
not, even the ones on the back that have the same power specs as the port
on the front. Generally I know the suggestion is reversed- plug into the
back closer to the motherboard- but for some reason this resolved the issue
for me. The extender cable I used here is not long enough to reach my
computer in the gallery, but I have ordered another that will arrive
tomorrow, and am happy to report back if that one ends up working.

Tested USB Extender:
https://www.amazon.com/gp/product/B07QDXJR9Z/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

Untested USB Extender:
https://www.amazon.com/gp/product/B06Y29FJSQ/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

Thank you for the prompt response! It would be a great resource to have a
list of tested extender cables that work for long distances, as even 32
feet is still quite limiting depending on the application and the cords
provided are very short. I'm assuming most people are covering long
distances in warehouses and such, so it would be helpful to know how to
accomplish this successfully. I appreciate your help.

wes-b, 2020-02-10:
It would be great if you could update the USB Extender topic with your findings.

For the port issue, I suspect your front USB ports connect to a different USB Host controller than the back. If you want we can open an issue for that I can share more info & explore that problem

xutianfen, 2021-10-04:
Hi
I am a new guy for Azure Kinect camera.
I met the same problem as above

OS with Version: Windows 10 Enterprise version 2016 LTSB
SDK Version: 1.4.1

After clicking the START button, I got the same logs as above.
I have already reinstall the windows but unfortunately the problem is still there.

BTW, the camera works well when it is connected to another PC with 
Windows 10 Pro.

Do you have any advice on what I can do next?