microsoft/Azure-Kinect-Sensor-SDK #1036: Body tracking program seems work without using GPU

xiongmao4hao, 2020-01-17:
I'm using Azure Kinect Body Tracking SDK 0.9.5. 
I tested the official body tracker executable file k4abt_simple_3d_viewer.exe. It works well and my 2080Ti graphic card is running according to the task manager.
Then, I wrote my own code of body tracking, and I thought I did all the correct GPU configurations in my code. However, although my program works, it's very slow, and it seems that my 2080Ti didn't work.
Is there anyone who knows the possible problems or the solution? Thanks!

QuiqueMartinez, 2020-01-17:
It happened also to me. Actually I would says that k4abt_simple_3d_viewer.exe is a little bit slower than the 0.9.4 version.

yijiew, 2020-01-17:
@xiongmao4hao Here is the code building the official body tracker executable. Could you test whether it has the same issue? https://github.com/microsoft/Azure-Kinect-Samples/tree/master/body-tracking-samples/simple_3d_viewer

Specifically, are you using the cpp API or the C API for your own application? For the C API, you need to initilaize your tracker config to the default struct: k4abt_tracker_configuration_t tracker_config = K4ABT_TRACKER_CONFIG_DEFAULT.

QuiqueMartinez, 2020-01-21:
What I see is when using both versions of the tool (executables). I
recorded two videos with my mobile phon hitting the table with my hand and
counting the frames between the sound and the image reaching the table. In
0.9.5 time was clearly longer.

yijiew, 2020-01-23:
@QuiqueMartinez Thanks for conducting the test. The issue you mentioned seems to be a different issue than what this post is about. The thing you observe might be an algorithm latency issue instead of the configuration issue. Do you mind creating a separate GitHub Issue about the issue you mentioned?

yijiew, 2020-01-23:
@xiongmao4hao Was you problem resolved?

xiongmao4hao, 2020-01-24:
I am sorry to get back to you so late. Finally, I and my classmate find this issue is caused by the wrong dynamic link library which does not suit the version of the SDK on my computer. It is from the low version SDK, and we forget to replace it in my own application.

closed 2020-01-24 by xiongmao4hao