microsoft/Azure-Kinect-Sensor-SDK #1037: Table adds to the segmentation when arm is in contact.

QuiqueMartinez, 2020-01-17:
Hi, when the user's arm is in contact with the table there are artifacts in the segmentation as clearly can be seen in the viewer. It should be pretty straightforward to remove portions of the segmentation that due to their curvature cannot be part of a human body.

I wanted to share something similar I did for the Kinetc2 some time ago. I apologize for the quality of the debug, but in the video, I'm trying to find surfaces in which curvature is coherent with a human body, discarding flat surfaces. I hope it could be useful.

[https://www.youtube.com/watch?v=Dt9waXZ0S3Y](url)

qm13, 2020-01-17:
@QuiqueMartinez thanks for the feedback. Improving instance segmentation is one of the areas we plan to invest post v1.

qm13, 2021-03-24:
Duplicate of #949.

closed 2021-03-24 by qm13