microsoft/Azure-Kinect-Sensor-SDK #1039: Sensor orientation and expected results

JZharay, 2020-01-17:
So this problem was something I wasn't expecting when I was implementing more features into our applications. When you configure the body tracker to have a sensor orientation of 90 degrees or even upside down, I expected the result to maintain positions and orientations in a normal XYZ axis as you already have established. However, the resulting body is rotated along with the sensor as well.

So for example: say you are sitting in front of it, T-Posing. In default orientation and the camera in a default orientation, you will be properly up-right and dominating your simulated space. However, If you set the sensor to 90 clockwise both physically and in the configuration, you will now be T-Posing 90 degrees to the side where left is up and right is down.

I expected my dominating pose to remain as if I hadn't rotated my 3D avatar but I I guess that is not the case? Was this as intended or was this supposed to do as I had envisioned?

**Setup Details**
- Azure Kinect SDK: v1.3.0
- Azure Kinect Body Tracking: v0.9.5

**Side note**
Previously you had an error in which you had the orientation configuration not working correctly (fixed in 0.9.3), but in this case tracking is working perfectly fine. Its just the final orientations I am not satisfied with that brings me here.

yijiew, 2020-01-17:
@JZharay Thanks for raising this question. Actually, it is intended. Because all the positions and orientations are relative to the depth sensor coordinate system instead of the real world coordinate system. You can refer here for the detailed definition of the sensor coordinate system: https://docs.microsoft.com/en-us/azure/kinect-dk/coordinate-systems. 
For your application, since you know the orientation of the sensor, it should be easy to apply coordinate transformation to make them up right.

JZharay, 2020-01-19:
That kinda sucks. I'll have to swap coordinates then and make them work for when we choose to go counter-clockwise (upside-down would mean just making everything go negative, not so bad).

Thanks for the fast reply @yijiew !

closed 2020-01-19 by JZharay

JZharay, 2020-01-27:
Just to add a bit of info for those of you who are working with this problem. Know that your avatar's pelvis is your pivot point. Easiest way to get around this issue is to simply rotate the whole thing by that point.

ARportals, 2020-08-14:
I agree with JZharay that I was expecting the body to remain at the same orientation/position independently of the sensor's orientation. Maybe it's expected behavior to Microsoft but to every other user, it's not expected. Why would I have to change my software when changing the orientation or a sensor? Doesn't make sense to me

ARportals, 2020-08-14:
@JZharay @yijiew How would I go about applying the new coordinates? I tried rotating the body but then the body is floating around even though it's now straight...

ARportals, 2020-08-15:
Just figured it out. The easiest way to make sure your characters stay at a proper angle is to apply a rotation offset based on the device orientation setting. So for every body that was found, then for every joint, you have to apply that rotation offset on the incoming joints rotation AND position. Every joints position and rotation are in absolute so you have to apply the rotation offset to every joint otherwise it won't work.

ARportals, 2020-08-15:
Of course, if you have converted your position and rotation in relative mode then you could only apply the rotation offset to the position and rotation of the waist joint like mentioned before.