microsoft/Azure-Kinect-Sensor-SDK #1041: undistort glsl pixel shader help.

barakooda, 2020-01-19:
Trying to write simple undistort glsl pixel shader implementing [ Brown-Conrady Distortion Model.](http://www.imatest.com/docs/distortion-models/)
I`m getting kind of opposite result ( more distortion) .

Any tips in how to use the coefficients and build the polynom ? 
Thanks!

barakooda, 2020-01-20:
trying different ways, 
this time i`m looking here :  
I`m not using the right units... i`m kind og guessing what i have to do with the
k4a intrinsic data ... 

[image]
 

https://docs.opencv.org/2.4/modules/imgproc/doc/geometric_transformations.html#void%20initUndistortRectifyMap(InputArray%20cameraMatrix,%20InputArray%20distCoeffs,%20InputArray%20R,%20InputArray%20newCameraMatrix,%20Size%20size,%20int%20m1type,%20OutputArray%20map1,%20OutputArray%20map2)

`
// principal point x and y , focal length x and y , all calculated in pixels 
uniform vec4 cameraMatrix;

//k1,k2,k3,k4
uniform vec4 disto1;

//k5,k6,tangential distortion coefficient x and y
uniform vec4 disto2;

out vec4 fragColor;

void main()
{
	
	vec2 res = uTD2DInfos[0].res.zw;
	
	vec2 PrincipalPoint = cameraMatrix.xy;
	vec2 focalPoint = cameraMatrix.zw;
	
	
	
	
	vec2 uv = vUV.st * res ;
//---------

	vec2 finalUV = vec2(0);

	vec2 xy = (uv - PrincipalPoint) / focalPoint;
	
	//r power 2
	float rp2 =  pow(xy.x,2) + pow(xy.y,2);
	

	float distoK = (1 + 
	disto1.x * rp2 + 
	disto1.y * pow(rp2,2) + 
	disto1.z * pow(rp2,3) +
	disto1.w * pow(rp2,4) +
	disto2.x * pow(rp2,5) +
	disto2.y * pow(rp2,6)

	);
	
	
	float distoTanX = 2 * disto2.z * xy.x * xy.y + disto2.w * (rp2 + 2 * pow(xy.x,2) );
	
	float distoTanY =  disto2.z * (rp2 + 2 * pow(xy.y,2) + 2 * disto2.w * xy.x * xy.y );
	
	finalUV.x = uv.x  * distoK  + distoTanX ;
	finalUV.y = uv.y  * distoK  + distoTanY ;
	
	finalUV = finalUV * focalPoint + PrincipalPoint;
	
	
//----------	
	finalUV = finalUV / res;

	//vec4 color = texture(sTD2DInputs[0], finalUV);
	vec4 color = vec4(finalUV,0,1.0);
	fragColor = TDOutputSwizzle(color);
	
}
`

barakooda, 2020-01-21:
**Current status** 

Undistorted image using opencv undistort() function : 
[image]

glsl undistort :
[image]

difference between glsl and opencv undistort : 

[image]

rabbitdaxi, 2020-01-22:
@barakooda regarding the difference between yours and opencv (by just looking at the images above), have you checked what interpolation method you are using for opencv remap? based on your shader code, it seems equivalent to nearest neighbor, do you use LINEAR or NN for the opencv remap during undistortion? 

Not related to opencv, we posted an example https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/examples/undistort/main.cpp that you can refer to. You can change how to choose the pinhole based on your need.

barakooda, 2020-01-22:
Hi, thanks for looking on my code,

> what interpolation method you are using for opencv remap?
"remap( src, dst_part, map1_part, map2_part, INTER_LINEAR, BORDER_CONSTANT );"
Based on the opencv [undistort function](https://docs.opencv.org/4.1.2/d9/d0c/group__calib3d.html#ga69f2545a8b62a6b0fc2ee060dc30559d)
The remap uses bilinear interpolation.

The shader runs in touchdesigner and the textures defined with GL_LINEAR mode
 which means bilinear filtering.

> Not related to opencv, we posted an example
ultimately i should compare my ouput to kinect4a method,
I guess it will be better reference no ?

barakooda, 2020-01-23:
Update:
I will have to do more tests but as it seem,
the problem was the Y axis difference between GL and opencv.

so everything seem to work well.
thanks again.

closed 2020-01-23 by tesych

oUp2Uo, 2023-03-31:
Hi @barakooda, 
I am trying to write a undistortion glsl program and found this topic.

I tried to write the formula here:
http://opencv.jp/opencv-2.1_org/py/camera_calibration_and_3d_reconstruction.html#initundistortrectifymap
Just similar to your code.
But I could not get the correct result. 
I am calculating new uv coordinate in vertex shader, but the result is not the same as code running on CPU.
```
			uniform dvec4 disto1;
			uniform dvec4 disto2;
			uniform dvec2 focal;//focal
			uniform dvec2 uv;//center
			uniform ivec2 image_size;
```
```
				double x = m_p.x / m_p.z;
				double y = m_p.y / m_p.z;

				double x2 = x * x;
				double y2 = y * y;
				double r2 = x2 + y2;
				double m_2xy = 2 * x * y;

				double k1 = disto1.x;
				double k2 = disto1.y;
				double p1 = disto1.z;
				double p2 = disto1.w;
				double k3 = disto2.x;
				double k4 = disto2.y;
				double k5 = disto2.z;
				double k6 = disto2.w;

				double kr = (1.0lf + ((k3 * r2 + k2) * r2 + k1) * r2) / (1.0lf + ((k6 * r2 + k5) * r2 + k4) * r2);
				double u = focal.x * (x * kr + p1 * m_2xy + p2 * (r2 + 2 * x2)) + uv.x;
				double v = focal.y * (y * kr + p1 * (r2 + 2 * y2) + p2 * m_2xy) + uv.y;
```
Have you ever met problem like this?
Thanks.