microsoft/Azure-Kinect-Sensor-SDK #1042: How to get/output the distance a person is away from the Kinect once their body is detected?

MaaaChu, 2020-01-20:
I'm pretty new to developing with the Kinect and I'm using the Kinect to extract data a person's movements in a room. 

Firstly, I want to get the distance a person is away from the Kinect after their body has been detected using the body tracking SDK.
I'm looking to achieve something similar to the Cognitive Services example where a distance in mm is output when a person is detected.

Any suggestions are appreciated! Thanks.

RoseFlunder, 2020-01-20:
My guess would be choosing a position from the body (i.e. head) and just calculate the euclidean distance to 0|0|0 because thats the position of the depth camera.

Example:
Head Position is at 2|1|1:
Distance = sqrt((2-0)²+(1-0)²+(1-0)²)

MaaaChu, 2020-01-21:
Thanks, @RoseFlunder.

I'm using the values generated by the offline processor example application. Have you used this? If so, do you have any understanding of what the values generated relate too? E.g the joint position values, are they referring to x,y, z co-ords or distance?

closed 2020-01-21 by rabbitdaxi

Sagid28, 2020-02-12:
> My guess would be choosing a position from the body (i.e. head) and just calculate the euclidean distance to 0|0|0 because thats the position of the depth camera.
> 
> Example:
> Head Position is at 2|1|1:
> Distance = sqrt((2-0)²+(1-0)²+(1-0)²)

can you give me the source code please if you have.

RoseFlunder, 2020-02-12:
Sorry I don't have an working example at hand, but you could check out this example on how to get all the joint positions:
https://github.com/microsoft/Azure-Kinect-Samples/blob/master/body-tracking-samples/simple_cpp_sample/main.cpp#L15

So after line 15 you would calculate the joints distance to the camera like this:
`sqrt(position.v[0]*position.v[0] + position.v[1] * position.v[1] + position.v[2] + position.v[2])`

For the "sqrt" function you would need to include "math.h"

angelicaCruz, 2023-01-11:
Piggy backing on this comment, is there a way to get the distance of a segmented body that does not have body tracker active on it?