microsoft/Azure-Kinect-Sensor-SDK #1044: random k4abt.dll crash

neilDGD, 2020-01-21:
lib versions: k4a sdk version 1.3.0, Body tracking sdk version 0.9.5.
OS version: Windows 10 Pro Insider Preview version 19541.1000.
Device firmware versions: RGB camera 1.6.102, Depth camera 1.6.75, Audio 1.6.14.

Hi,

We get random crashes within k4abt.dll with no indication what the error was (after a length of time of using the library and tracking a single body perfectly fine). We get an access violation somewhere inside k4abt.dll:

Unhandled exception at 0x00007FFB589D4E27 (k4abt.dll) in FX_d.exe: 0xC0000005: Access violation reading location 0x000000045CB03D90.

The stack is just a bunch of addresses in k4abt.dll with no context.

as the pdb's aren't distributed with the dll's we are unable to pinpoint the reason / location of the crash.

The test is literally just tracking one body for a length of time, not much happening just waving arms about occasionally, the state is pretty much the same for the duration, i.e. same amount of joints in frustum.

Would be greatful to hear if this is known and has a fix pending?

Thanks.

qm13, 2020-01-27:
We are unable to repro this issue. We have a number of body tracking demo stations running v0.9.5 body tracking viewer for over 1 month with no crashes. More recently we have been running v1 final build for over 1 week with no crashes. Could you please attempt to run the body tracking viewer on your system for an extended period of time.

neilDGD, 2020-01-27:
I just tested by running k4abt_simple_3d_viewer.exe, and the app crashes after 2 minutes here. Ofcourse I cannot debug, if you have a version of the viewer test app that is useful for narrowing down the reason - maybe with logging to file, that would be most helpful.
I have run our app with full k4a logging enabled and attached the log, this may be helpful to you?
[kinectlog_000036592212.txt](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/files/4118847/kinectlog_000036592212.txt)

yijiew, 2020-01-27:
@neiDGD the log doesn't seem to give us any useful information. It seems that you enabled the k4a logging. Have you enabled the k4abt logging?

neilDGD, 2020-01-27:
can you point me to the api to enable k4abt logging? There does not seem to be one from my search.

neilDGD, 2020-01-27:
Ok I enabled logging using environment variables (with info level logging).  When I log to a file, it accepts my path but doesn't produce the file, so fell back to console logging and all I got before k4abt_simple_3d_viewer.exe crashes is:

**[2020-01-27 13:58:12.392] [info] [t=12020] [K4ABT] D:\a\1\s\src\TrackerHost\TrackerHost.cpp (465): VerifyDependencies(). Verifying Dependencies!
[2020-01-27 13:58:12.393] [info] [t=12020] [K4ABT] D:\a\1\s\src\TrackerHost\TrackerHost.cpp (146): Create(). Initializing CUDA and ONNX Runtime...
[2020-01-27 13:58:15.000] [info] [t=12020] [K4ABT] D:\a\1\s\src\TrackerHost\TrackerHost.cpp (168): Create(). Done.**

Any other ideas?

neilDGD, 2020-01-27:
[image]

Could my host controller type be causing k4abt to fail after some time? doesn't seem likely to me?

wes-b, 2020-01-27:
> Could my host controller type be causing k4abt to fail after some time? doesn't seem likely to me?

We don't have a lot of data on AMD controllers, Intel and TI seem to work the best.  If you have an expansion card installed and have not properly hooked up the supplemental power, then you may run into issues.

neilDGD, 2020-01-27:
Just a sanity check; it is ok to call the k4bt tracker::enqueue_capture(), tracker::pop_result(), frame::get_body() etc from a separate thread right?

We capture any thrown exceptions during frame retrieval using those calls and nothing is throwing.

yijiew, 2020-01-27:
Thanks for enabling the log. The log looks fine without problem... And the API doesn't throw exceptions. It shouldn't be the problem of multi-threading, because you mentioned you have tried "running k4abt_simple_3d_viewer.exe, and the app crashes after 2 minutes here". The k4abt_simple_3d_viewer.exe is single-threaded.

closed 2020-02-06 by qm13

Zoed1011, 2024-05-22:
[CrashContext.runtime-xml.txt](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/files/15397662/CrashContext.runtime-xml.txt)
lib versions: k4a sdk version 1.4.1, Body tracking sdk version 1.1.2.

Hi,

I am encountering the same issue. While developing my game using the Body Tracking SDK, everything runs smoothly initially, and the game operates as expected. However, after a certain period, the application suddenly crashes without any warning. During this time, I am not performing any specific actions, just waving my hands and moving around in front of the camera.

The timing of the crashes is very unpredictable. Sometimes the application can run for an entire day without crashing, while other times it crashes multiple times within just 30 minutes. Each crash report indicates that the issue occurs within the stack of k4abt.dll, but I cannot determine the exact cause. I have carefully reviewed many documents and used the SDK with caution. Every critical API is protected with try-catch blocks, but the crashes still occur.

I would like to confirm if you have resolved this issue and, if so, how you managed to do it? During this period, I have tried various methods, including testing with different DLLs, ONNX files, and both single and multi-threading, but the same random crashes persist. I would appreciate any insights or solutions to this issue.

Please let me know if you need any further information from me. Thank you very much.