microsoft/Azure-Kinect-Sensor-SDK #1047: when calling k4a_device_stop_cameras() on shutdown, errors occur and unable to cleanly shutdown app.

neilDGD, 2020-01-25:
lib versions: k4a sdk version 1.3.0, Body tracking sdk version 0.9.5.
OS version: Windows 10 Pro Insider Preview version 19541.1000.
Device firmware versions: RGB camera 1.6.102, Depth camera 1.6.75, Audio 1.6.14.

Hi,

When calling k4a_device_stop_cameras() on shutdown, we get the following logged errors:

System\Kinect4AzureManager.cpp(124): WARNING: k4alib (error | critical): **usb_cmd_io(). Error calling libusb_bulk_transfer for initial tx, result:LIBUSB_ERROR_OTHER**
System\Kinect4AzureManager.cpp(124): WARNING: k4alib (error | critical): **usb_cmd_io(usbcmd_handle, cmd, p_cmd_data, cmd_data_size, NULL, 0, p_data, data_size, NULL, cmd_status) returned failure in usb_cmd_write_with_status()**
System\Kinect4AzureManager.cpp(124): WARNING: k4alib (error | critical): **usb_cmd_write_with_status(depthmcu->usb_cmd, DEV_CMD_DEPTH_STREAM_STOP, NULL, 0, NULL, 0, &cmd_status) returned failure in depthmcu_depth_stop_streaming()**
System\Kinect4AzureManager.cpp(124): WARNING: k4alib (error | critical): **usb_cmd_io(). Error calling libusb_bulk_transfer for initial tx, result:LIBUSB_ERROR_OTHER**
System\Kinect4AzureManager.cpp(124): WARNING: k4alib (error | critical): **usb_cmd_io(usbcmd_handle, cmd, p_cmd_data, cmd_data_size, NULL, 0, p_data, data_size, NULL, cmd_status) returned failure in usb_cmd_write_with_status()**
System\Kinect4AzureManager.cpp(124): WARNING: k4alib (error | critical): **usb_cmd_write_with_status(depthmcu->usb_cmd, DEV_CMD_DEPTH_STOP, NULL, 0, NULL, 0, &cmd_status) returned failure in depthmcu_depth_stop_streaming()**
System\Kinect4AzureManager.cpp(124): WARNING: k4alib (error | critical): **Stop(). Failed to request flush for stop: 0xc00d3e85**

I would be grateful for any insights or confirmation this is a known problem or the cause, and whether a possible fix would be available in the future if it is a known problem?

I have attached a full log with all categories of log entry included for a run of our app.

Many thanks,

Neil
[kinectlog_000036592212.txt](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/files/4111521/kinectlog_000036592212.txt)

wes-b, 2020-01-27:
I can't reproduce this myself. Can you reproduce this with our version of k4aviewer (from our MSI package)? 

For the logs it would be helpful to capture the thread ID and timestamps. If you do collect logs again please look at https://docs.microsoft.com/en-us/azure/Kinect-dk/troubleshooting#collecting-logs.

I would love to know what the process call stacks are when ```queue_disable(). Queue "dewrapper" waiting for blocking call to complete.``` is repeating. Can you catch this on a debugger?

neilDGD, 2020-01-27:
ok, i enabled logging using env vars, and I don't get a log file out at all. As per my other k4abt bug post, i added env vars for path log level and no file out. I see logging like this is the only way to check the k4aviewer app (as it occurs on shutdown in our app). Any ideas why the log file wouldn't be output?

callstack of the "info" level log event for **"queue_disable() Queue "dewrapper" waiting for blocking call to complete."** is:

[image]

neilDGD, 2020-01-27:
[k4a.log](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/files/4119288/k4a.log)

Ok got the log working ("K4A_ENABLE_LOG_TO_A_FILE - ** When enabled this takes precedence over the value of K4A_ENABLE_LOG_TO_STDOUT" isnt true it appears), attached the log. Although nothing seems out of the ordinary running the k4aviewer app. So, in this case we can rule out the usb controller. Could it be some sort of threading issue?

I also had a Kinect2 attached as we support many types of sensor, so i detached that but the same resulting log happened, I also tried a 3.1gen2 usb port, but that drops frames crazy anyway - and same resulting log output.

wes-b, 2020-01-27:
The log doesn't appear to be a repro of the same issue, there are no errors in it. 

The callstack you captured is what I expected. Unfortunately I was asking for the other thread callstacks to know who might have been holding the lock that is needed for waiting thread to proceed.

neilDGD, 2020-01-28:
The only thread active at the point of calling Device::stop_cameras() is the main thread.
[image]
Our K4A Manager object is being destructed globally as it is a singleton instance, are there any problems with this?

wes-b, 2020-01-28:
Darn, that isn't right, any chance you can share your code?

neilDGD, 2020-01-29:
Hey, this was a bit of a goose chase, it _was_ to do with global destruction on DLL Detachment. When closing devices before global destruction it works now. Thanks for the patient support.

closed 2020-01-29 by neilDGD

wes-b, 2020-01-29:
Hmm, I would still be interested in the repro. You were stuck in a loop that wasn't exitable, and that you should not have been in.