microsoft/Azure-Kinect-Sensor-SDK #1051: Is the body tracking model closed-source?

slimsag, 2020-01-30:
Hello -- I was unable to locate the ML code and training data that the Azure Kinect uses for body tracking. I am wondering if it is closed-source or not, and if so, whether or not there are plans to release it?

I am interested in full-hand pose and facial expression pose estimation for the application of 3D modeling, and would like to try using the body tracking implementation as a reference if possible.

qm13, 2020-02-06:
Yes, body tracking is closed source.

closed 2020-02-06 by qm13

eshirima, 2020-03-02:
@qm13 I was testing out the body tracker and realised that in most of the [samples](https://docs.microsoft.com/en-us/azure/kinect-dk/build-first-body-app), the colour camera is turnt off. Is it safe to assume that the body tracker only utilises the depth camera for detection & tracking?

qm13, 2020-03-02:
Yes, body tracking only uses the depth and IR images. RGB camera is not used.