microsoft/Azure-Kinect-Sensor-SDK #1052: multi-trackers in 1PC, 1GPU?

lyw96, 2020-01-30:
I found that v0.9.5 of the Body Tracking SDK provides multi-tracker support to track bodies from different Azure Kinect devices in 1 PC.
I wonder if the multiple trackers require multiple GPUs, or it is possible to use 1 GPU(I'm using RTX 2080) for multi-trackers. I could not find any explanations for use 1 GPU for multi-trackers.

And I have one more question. Could the CPU only mode for running the tracker handle multi-trackers?

qm13, 2020-02-06:
You can run multiple trackers on a single GPU.

closed 2020-02-06 by qm13

NightChord, 2020-06-09:
"I found that v0.9.5 of the Body Tracking SDK provides multi-tracker support to track bodies from different Azure Kinect devices in 1 PC."
I would like to ask, can Body Tracking of multiple kinects be integrated? Has this api? Please tell me how to merge multiple kinect Body tracking data.

jesusatgar, 2020-12-02:
Any docs on this?