microsoft/Azure-Kinect-Sensor-SDK #1053: Hand state classifier and fully articulated hand tracker

rfilkov, 2020-01-30:
**Is your feature request related to a problem? Please describe.**

Hi,

This is the most requested body tracking feature since months now (apart from MacOS and ARM support): [](https://feedback.azure.com/forums/920053-azure-kinect-dk/suggestions/38933653-fully-articulated-hand-tracking-and-classifiers-fo)

It stays in 'Need feedback' state since a long time. Nobody replies to the user comments, too. The user experience in the feedback forum is not great overall. The only recent change is that the feature request was renamed that way. Until then it was only a request for hand state classifier.

**Describe the solution you'd like**

Do you plan to add any of these features to the Body Tracking SDK any time soon? 

**Describe alternatives you've considered**

**Additional context**

RobJellinghaus, 2020-01-31:
+1. Though a classifier and a full tracker are two different things, and the issue was originally specifically about a classifier, which is a lot less work than a full tracker. So I am not sure about the wisdom of changing the issue title after 70+ votes had already occurred.  A full tracker would be great, but there is lots of interest in just a Kinect v2 style classifier.

closed 2020-06-27 by rfilkov

RobJellinghaus, 2020-06-30:
@rfilkov why did you close this? Did it get implemented?

rfilkov, 2020-06-30:
@RobJellinghaus There is nothing new implemented in the body tracking since v1.0. My patience has some limits.

vongomben, 2021-05-24:
Dop you think this is going to be implemented soon? 
The hancd traking is now reaching the hand tip 
https://microsoft.github.io/Azure-Kinect-Body-Tracking/release/1.1.x/group__btenums.html