microsoft/Azure-Kinect-Sensor-SDK #1055: Multi-device auto-exposure

SeanStephensenAIRM, 2020-01-31:
This is either a suggestion, or a request for guidance if this is already possible. We're looking at a daisy-chained set of 2+ devices for long-term automated capture in a setting with changing lighting conditions. My concern here is the need for manual exposure setting to avoid the issue of auto exposure leading to different exposures across devices. Since our lighting may be changing, it would be idea for me to set the master camera to auto-exposure (and I would place the master in an optimized spot for exposure-detection), and then share that same exposure value to each subordinate device. Without this function, I believe I have to manually determine exposure settings for each lighting condition and use manual-exposure-control. Best case scenario would be a feature that doesn't require re-starting the cameras (as I believe manual exposure must be set before camera start?): At any point during operation (just like live streaming), master camera should detect auto-exposure and return that value, which can be used to "manually" set sub camera exposures. 
Furthermore, can someone clarify the following points from [this page](https://docs.microsoft.com/en-us/azure/Kinect-dk/multi-camera-sync):

- Do not repeatedly set the same exposure setting in the image capture loop

- Do set the exposure when needed, just call the API once

Thanks

tesych, 2020-02-03:
@SeanStephensenAIRM, you can do it today also by setting the auto- exposure for master and programmatically adjust the subordinates.. However, you may loose some frames when the exposure adjustment happens.

SeanStephensenAIRM, 2020-02-04:
@tesych thanks for the reply. Can you elaborate on how to programmatically adjust the subordinates to match the master's auto-detected exposure? Losing frames is not an issue for us - we only need one capture every 30 seconds or so.

tesych, 2020-02-04:
@SeanStephensenAIRM, you would need to call [k4a_image_get_exposure_usec()](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/group___functions_ga54568ebab477f2534b3c4c92688344d9.html#ga54568ebab477f2534b3c4c92688344d9) to get the exposure from the master device and compare to the last value you set for subordinates. Then set the subordinate devices to the new value by calling [k4a_device_set_color_control()](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/group___functions_gae81269489170b26b7f0bbe1a7f9d31d6.html#gae81269489170b26b7f0bbe1a7f9d31d6).
Also, if losing frames is not an issue for you and you do the capture every 30 sec, I do not believe it would cause any problems for your scenario to set all of the cameras to auto- exposure.

SeanStephensenAIRM, 2020-02-04:
@tesych Thanks. To clarify, it seems like k4a_image_get_exposure_usec() is called on an image that is extracted from a capture. So in theory, to get a synchronized capture, would I need to take an auto-expose capture with the master cam, store the exposure value from that image, and then use that value to take a **manually** exposed capture from both cameras? The simpler alternative would be to keep the images from the auto-exposed master capture, and have a small delay before the manually exposed sub capture. Not sure how much that delay would be, it might be well within the 160us range anyways.
This still leaves some uncertainty with my last question regarding IR interference. If this process of master capture -> master image -> get_exposure -> reconfigure sub -> sub capture has any delay, especially if it is variable, how can I be sure that IR interference will not occur since IR delay is set off color capture for each device, as in the green screen example? Would I need to find that new delay time caused by reconfiguration and factor that into my manually set delay time for IR?

tesych, 2020-02-05:
@SeanStephensenAIRM, the auto- exposure settings should not affect the depth camera. But, let's step back and try to understand what exactly you are trying to achive?

SeanStephensenAIRM, 2020-02-05:
@tesych Once every period>=30s, I would like to capture a time synchronized set of auto-exposed images from some number of (2 for now, maybe more later) cameras. This occurs under changing lighting spectra and intensities.

Like you've mentioned, I think for now, we will simply auto-expose all cameras and correct any difference in post-normalization, but I'd like to understand the process for how manual exposure may apply to this scenario.

wes-b, 2020-02-05:
What accuracy do you need for the synchronized images? I.E. Camera A image should be within ?ms of Camera B image. Auto exposure can go from 500us to over 100ms when running at 5FPS

If you only want an image every 30s then it seems like you could poll each camera one at a time.

SeanStephensenAIRM, 2020-02-05:
@wes-b Thanks for the reply. For our current use, low sync accuracy is required. You're right, I could work around any of this confusion by setting each camera to capture 1 second apart and not have to think about the IR interference at all. I guess now I'm just curious in case we move to higher frequency images in the future. That slightly unpredictable ~500us+ delay is what concerns me if auto exposure is used. If there is going to be some somewhat unknown delay, is there any way of ensuring that IR interference does not occur? Since the delay_off_color_camera which prevents interference is no longer precise if it's added to some other unknown delay. 

Is this why the green screen example code states "you need to manually set both their exposures"?

If that is the case, then I believe we have decided that the following options apply to a dynamic lighting scenario with looping image capture:

1. Use auto exposure on all cameras and lose precise control over IR interference prevention. Set each camera capture far enough apart to guarantee non-overlapping depth captures, or run the risk of IR interference.

2. Use auto exposure on just the master camera. Fetch that exposure value and manually re-configure all subordinate cameras before obtaining synchronized captures from those sub-cameras, which will be slightly delayed from the master image.

3. Option 2, but after fetching the auto-detected exposure value, reconfigure all cameras, including master. Obtain synchronized images from all cameras. Exposure detection/configuration takes longer, but all images are synchronized with IR interference prevented.

If this is correct, thank you both for your advice, and the issue may be closed.

wes-b, 2020-02-06:
The simplest solution is to ensure you don't have depth overlap. You might even experiment with how bad the interference is. It may not be that bad.

Your #2 with auto exposure on master seems like the next best option. But that relies on all other camera's needing the exact same exposure. That may or may not be realistic for your application. It definitely won't work for everyone.

The most difficult synchronization happens when each camera has its own and unique exposure setting. In this scenario you will need to redefine the master and subordinate order. I think from longest exposure to smallest, but I have to look that up

closed 2020-02-20 by wes-b