microsoft/Azure-Kinect-Sensor-SDK #1056: Suggest to fix API Reference

UnaNancyOwen, 2020-02-03:
**Is your feature request related to a problem? Please describe.**

Currently, API Reference has the following problems.

* Currently, C# Wrapper API is treated as Data Structures.
  I think it should be treated as one language such as C, and C++.
  [image]

* Azure Kinect Body Tracking SDK doesn't have API Reference for C # Wrapper.
  It has already been implemented from 0.9.5. I think it should be documented.
  [image]

**Describe the solution you'd like**

Add and Fix the API reference of C # wrapper appropriately.

**Describe alternatives you've considered**

**Additional context**

tesych, 2020-02-03:
@UnaNancyOwen thank you for the suggestion. We already have the issue opened for this #616. Closing as duplicate.

closed 2020-02-03 by tesych

UnaNancyOwen, 2020-02-04:
@tesych Thanks, Don't forget to add C# wrapper API reference to Azure Kinect Body Tracking SDK document!