microsoft/Azure-Kinect-Sensor-SDK #1057: The PDB files in the Body Tracking SDK are missing

rfilkov, 2020-02-03:
**Describe the bug**

The PDB-files of the native libraries in Body Tracking SDK are missing. This makes it difficult to guess the source of an error, unless we decide to start diving into the Assembler code.

For instance, this happens on some machines, causing a crash, when invoking k4abt_tracker_create():
"
0x00007FF86B873E39 (KERNELBASE) RaiseException
0x00007FF8121CAFEB (k4abt) k4abt_tracker_destroy
0x00007FF8121C83A6 (k4abt) k4abt_tracker_destroy
0x00007FF8121B8AE5 (k4abt) k4abt_tracker_destroy
0x00007FF8121B0D28 (k4abt) k4abt_tracker_destroy
0x00007FF81202C265 (k4abt) k4abt_tracker_destroy
0x00007FF812037F19 (k4abt) k4abt_tracker_destroy
0x00007FF811FFC0F2 (k4abt) k4abt_tracker_destroy
"

**To Reproduce**

It will be difficult to reproduce each and every error at your end.

**Expected behavior**

Please add the PDB-files for 'k4abt.dll' (and eventually for 'onnxruntime.dll') to the future releases of 'Azure Kinect Body Tracking SDK'. This may help debug and localize the errors.

**Logs**

**Screenshots**

**Desktop (please complete the following information):**
 - OS with Version: Windows 1909
 - SDK Version:  Azure Kinect SDK 1.3.0, Body Tracking SDK 1.0.0

**Additional context**

yijiew, 2020-02-06:
@rfilkov Thanks for the feedback! As for your particular crash, could you share your crash dump with us so that we can take a look at the issue you hit? 
You can save a crash dump by following instruction here: https://docs.microsoft.com/en-us/visualstudio/debugger/using-dump-files?view=vs-2019.

rfilkov, 2020-02-10:
@yijiew This particular crash is the same as #1061 If you need a crash dump, we could ask Mike to create one for you. But as I see, you are pretty sure that it is due to a dll version mix-up of onnxruntime.dll. As a matter of fact, I've few crash reports from other customers before that too, but unfortunately don't keep their logs. My experience so far shows two main sources of crash - either at the time of initialization of depthengine_2_0.dll, or at the time of creation of the body tracker.

Anyways, this issue shows once again why PDB files (as some kind of debugging info) is needed. The call stack at the time of the crash (see above) doesn't provide any hint that it is somehow related to the loading of the onnx-runtime library, not to mention the exact reason of the crash. The log-file is empty. This makes it quite difficult to provide first line support to the customers.

yijiew, 2020-02-10:
Hi @rfilkov , thanks for raising this issue. We are discussing among our team about sharing the PDB files for the next release. Please stay tuned. 

But for this particular issue that caused your crash in this version, it would be very helpful if you could ask Mike to share the crash dump with us so that we could further investigate.

rfilkov, 2020-02-26:
Hi @yijiew , FYI: I have another bug report regarding a crash in the body tracking SDK. This time it happens incidentally while running, according to the user. Here is the call stack:

"
0x00007FFBF0DD6D99 (k4abt) k4abt_tracker_destroy
0x00007FFBF0DC4075 (k4abt) k4abt_tracker_destroy
0x00007FFBF0DC3824 (k4abt) k4abt_tracker_destroy
0x00007FFBF0C6EE6B (k4abt) k4abt_tracker_destroy
0x00007FFBF0C5DDE5 (k4abt) k4abt_tracker_destroy
0x00007FFBF0C5E2B5 (k4abt) k4abt_tracker_destroy
0x00007FFBF0C5EADA (k4abt) k4abt_tracker_destroy
0x00007FFBF0C3D966 (k4abt) k4abt_tracker_destroy
0x00007FFBF0D841E0 (k4abt) k4abt_tracker_destroy
0x00007FFC2FBE7BD4 (KERNEL32) BaseThreadInitThunk
0x00007FFC2FD6CED1 (ntdll) RtlUserThreadStart
"

yijiew, 2020-02-26:
Could  you send me a dump? You can save a crash dump by following instruction here: https://docs.microsoft.com/en-us/visualstudio/debugger/using-dump-files?view=vs-2019.

rfilkov, 2020-02-26:
OK, I asked the customer to create a crash dump, if possible. #1091 is his bug report here.

qm13, 2020-03-11:
Closed. Investigation covered in #1091

closed 2020-03-11 by qm13

rfilkov, 2020-03-12:
@qm13 Does closing this issue mean you're not going to provide PDB files for the BT SDK libraries, and all crash reports should be directed to you?

rfilkov, 2020-04-16:
https://feedback.azure.com/forums/920053-azure-kinect-dk/suggestions/40188511-please-replace-the-body-tracking-sdk-team