microsoft/Azure-Kinect-Sensor-SDK #1060: I want to distinguish the moment when my feet touch the ground

devLupin, 2020-02-04:
This problem may be too easy for other developers. No matter how I thought, I couldn't solve it.
Please help me.

I want to get the coordinates of the foot of the starting point and the foot of the arrival point when the foot is laid out.
I am thinking of the best way to distinguish this.

In short, the coordinates of the foot can be obtained, but they cannot be distinguished under specific conditions.

Please let me know if you have any sites or examples to help.
Thank you.

billpottle, 2020-02-06:
Are you starting on the ground? If so you could find the point when the y value is back where you started. Otherwise, have a look at the [ground plane detection example ](https://github.com/microsoft/Azure-Kinect-Samples/tree/master/body-tracking-samples/floor_detector_sample)

devLupin, 2020-02-07:
I think it's an example similar to what I was looking for.
Thank you for your help, I will ask again if you have any questions.

closed 2020-02-07 by devLupin