microsoft/Azure-Kinect-Sensor-SDK #1063: Get temperature through the IR camera

ClementVincent, 2020-02-06:
Hi all, 

I'm would like to develop a program able to lively detect faces and to check if it's a real person or just a picture/video  (in particular by liveliness, heat and 3D checking). 
To do so, I need to remotely measure the temperature of a body using the IR camera.
What I though about is :
- Get RGB image
- Detect face and crop the IR arround
- Compute the average temperature (+ maybe check on special coordinates)
So far, I'm not able to convert the a4k IR image into a more friendly structure. I don't really now either how to retrieve the temperature.
Is there a way to easily access this information through the SDK ?

I'm using the C# Wrapper, Emgu and Keras (for the Learning part)

tesych, 2020-02-10:
@ClementVincent, thank you for your question! This device was not designed to capture body temperature and unfortunately we do not have any plans to do so.

closed 2020-02-10 by tesych

Greendogo, 2020-02-11:
If I may add to this answer to give the asked some information they did not ask for but probably needs to hear. As far as I know, there is not a way to get radiometric information from any thermal image without proper radiometry which in most instances is built into a thermal sensor by means of a flat field shutter with a thermocouple attached. Otherwise you are not able to corollate the data to real world absolute temperatures.